immortalwrt/immortalwrt #642: [21.02] 内核日志出现问题，并且那个阶段无线不可用

asiaon123, 2022-03-22:
```
[32321.379413] mt76x2e 0000:02:00.0: MCU message 1f (seq 2) timed out
[32321.424590] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[32321.430095] mt76x2e 0000:02:00.0: Build: 1
[32321.434251] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[32321.455428] mt76x2e 0000:02:00.0: Firmware running!
[32321.463550] ieee80211 phy1: Hardware restart was requested
[32470.813954] mt76x2e 0000:02:00.0: MCU message 1f (seq 7) timed out
[32470.859066] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[32470.864570] mt76x2e 0000:02:00.0: Build: 1
[32470.868704] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[32470.889943] mt76x2e 0000:02:00.0: Firmware running!
[32470.898088] ieee80211 phy1: Hardware restart was requested
```

是HC5962

1715173329, 2022-03-22:
- if it works please just ignore it
- if you're using mt-drivers please switch to use mt76
- if mt76 doesn't work please report it to upstream openwrt

AmadeusGhost, 2022-03-22:
21.02 上的 mt7612 已经 EOL，要么 4.14 要么老毛子。

closed 2022-03-22 by AmadeusGhost

asiaon123, 2022-03-22:
> 21.02 上的 mt7612 已经 EOL，要么 4.14 要么老毛子。

好的，谢谢