immortalwrt/immortalwrt #644: autocore 更新之后编译失败

imoldchen, 2022-03-25:
4小时前的更新，同步之后编译，提示复制 ./files/generic/index.htm 时没找到文件

```
mkdir -p /home/ubuntu/build/immortalwrt/bin/targets/x86/64/packages /home/ubuntu/build/immortalwrt/build_dir/target-x86_64_musl/autocore-x86/autocore-2/ipkg-x86_64/autocore-x86/CONTROL /home/ubuntu/build/immortalwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /etc
cp -fpR ./files/generic/index.htm /etc/
cp: cannot stat './files/generic/index.htm': No such file or directory
make[3]: *** [Makefile:64: /home/ubuntu/build/immortalwrt/bin/packages/x86_64/base/autocore-x86_2-18_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/ubuntu/build/immortalwrt/package/emortal/autocore'
time: package/emortal/autocore/x86/compile#0.68#0.06#0.74
    ERROR: package/emortal/autocore failed to build (build variant: x86).
make[2]: *** [package/Makefile:116: package/emortal/autocore/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/build/immortalwrt'
make[1]: *** [package/Makefile:110: /home/ubuntu/build/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/build/immortalwrt'
make: *** [/home/ubuntu/build/immortalwrt/include/toplevel.mk:230: world] Error 2
```

jh0072110, 2022-03-26:
在9a5e242的fix之后依旧编译失败

msylgj, 2022-03-26:
> 在9a5e242的fix之后依旧编译失败（21.02）

错误日志贴下

jh0072110, 2022-03-26:
> > 在9a5e242的fix之后依旧编译失败（21.02）
> 
> 错误日志贴下
[11_Compile the firmware.txt](https://github.com/immortalwrt/immortalwrt/files/8355074/11_Compile.the.firmware.txt)

1715173329, 2022-03-26:
已修复。

closed 2022-03-26 by 1715173329