immortalwrt/immortalwrt #645: 编译失败，提示ERROR: modpost: "rt6_lookup" [/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/shortcut-fe/shortcut-fe-cm.ko] undefined!

helloyan, 2022-03-26:
编译action：https://github.com/helloyan/ipq4019-openwrt/tree/immortalwrt
主要错误：
2022-03-25T17:06:29.4070495Z   MODPOST /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/shortcut-fe/Module.symvers
2022-03-25T17:06:29.4140430Z ERROR: modpost: "rt6_lookup" [/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/shortcut-fe/shortcut-fe-cm.ko] undefined!
2022-03-25T17:06:29.4146377Z make[5]: *** [scripts/Makefile.modpost:124: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/shortcut-fe/Module.symvers] Error 1
2022-03-25T17:06:29.4147393Z make[5]: *** Deleting file '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/shortcut-fe/Module.symvers'
2022-03-25T17:06:29.4148869Z make[4]: *** [Makefile:1726: modules] Error 2
2022-03-25T17:06:29.4149524Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.10.107'
2022-03-25T17:06:29.4151557Z make[3]: *** [Makefile:84: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/shortcut-fe/.built] Error 2
2022-03-25T17:06:29.4152146Z make[3]: Leaving directory '/workdir/openwrt/package/kernel/shortcut-fe'
2022-03-25T17:06:29.4162592Z time: package/kernel/shortcut-fe/compile#4.77#0.63#5.39
2022-03-25T17:06:29.4167126Z     ERROR: package/kernel/shortcut-fe failed to build.
2022-03-25T17:06:29.4175232Z make[2]: *** [package/Makefile:116: package/kernel/shortcut-fe/compile] Error 1
2022-03-25T17:06:29.4189778Z make[2]: Leaving directory '/workdir/openwrt'
2022-03-25T17:06:29.4207274Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
2022-03-25T17:06:29.4208164Z make[1]: Leaving directory '/workdir/openwrt'
2022-03-25T17:06:29.4296802Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2022-03-25T17:06:29.4743639Z ##[error]Process completed with exit code 2.

helloyan, 2022-03-26:
[logs_5.zip](https://github.com/immortalwrt/immortalwrt/files/8354441/logs_5.zip)

1715173329, 2022-03-26:
不用 sfe 即可。

closed 2022-03-28 by helloyan