immortalwrt/immortalwrt #646: master 分支编译失败 已尝试单线也是失败

WXR2727, 2022-03-26:
make[2]: Leaving directory '/home/khor/immortalwrt'
package/Makefile:110: recipe for target '/home/khor/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/khor/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/khor/immortalwrt'
Build failed - please re-run with -j1 to see the real error message
/home/khor/immortalwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1

ksong008, 2022-03-26:
请帖j1的图

WXR2727, 2022-03-26:
VPS ubuntu 18 编译

WXR2727, 2022-03-26:
[image]

WXR2727, 2022-03-26:
Luci application 我通常只选passwall

1715173329, 2022-03-26:
不要用 master 分支。

WXR2727, 2022-03-26:
21.02 昨天编译一样问题 ， 今天下午编译的21.02就成功 可是老失联 R2S

WXR2727, 2022-03-26:
失联就是断网 后台进不去

WXR2727, 2022-03-26:
[image]
官方openwrt也是一样的问题

1715173329, 2022-03-26:
> 失联就是断网 后台进不去

体质问题，换驱动。rtl8152-vendor

closed 2022-03-26 by 1715173329

WXR2727, 2022-03-26:
大佬求教学 怎么换

WXR2727, 2022-03-27:
大佬驱动是bootloader么

AmadeusGhost, 2022-03-27:
> 失联就是断网 后台进不去

这边建议出掉 r2s 换个 x86 或者硬路由。