immortalwrt/immortalwrt #673: can't expand /overlay

LawssssCat, 2022-04-20:
I want to expand /overlay area and I tried following these instructions:
<https://openwrt.org/docs/guide-user/additional-software/extroot_configuration>
The operation of this instructions takes effect in lede firmware, but not in this firmware. 

Excuse me, do you know the reason for this?

AmadeusGhost, 2022-04-20:
> Excuse me, do you know the reason for this?

Which branch are you using? What about specific device? I guess maybe x86 device with ext4/f2fs overlay, currently fstools cannot handle this correctly.

LawssssCat, 2022-04-20:
> > Excuse me, do you know the reason for this?
> 
> Which branch are you using? What about specific device? I guess maybe x86 device with ext4/f2fs overlay, currently fstools cannot handle this correctly.

A:
branch 21.02 （and also also tried 19.7, expansion of /overlay is not effective）
x86 device with squashfs overlay,

---
Here are some details after I configure `/etc/config/fstab` and reboot

<html>
<body>

主机名 | ImmortalWrt
-- | --
型号 | VMware, Inc. VMware Virtual Platform
架构 | AMD Ryzen 9 3900X 12-Core Processor x 4C 4T (3799.988MHz)
目标平台 | x86/64
固件版本 | ImmortalWrt 21.02-SNAPSHOT r19294-91bdfd5474 / LuCI openwrt-21.02 branch git-22.106.63729-0b2ce44
内核版本 | 5.4.188

</body>
</html>

```bash
root@ImmortalWrt:~# block  info
/dev/loop0: UUID="d0f1f31c-68fd-4723-9201-84ecdb2d99a0" LABEL="rootfs_data" VERSION="1.14" MOUNT="/overlay" TYPE="f2fs"
/dev/sda1: UUID="4DF8-B35F" LABEL="kernel" VERSION="FAT16" MOUNT="/boot" TYPE="vfat"
/dev/sda2: UUID="52180f4e-66f311b9-5e10662a-11e42c6c" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/sdb1: UUID="7d3b4f5e-5dc4-415a-a9f8-7a0bd2cb8891" VERSION="1.0" TYPE="ext4"

root@ImmortalWrt:~# cat /etc/config/fstab

config global
        option anon_swap '0'
        option auto_swap '1'
        option auto_mount '1'
        option delay_root '5'
        option check_fs '0'
        option anon_mount '1'

config mount
        option target '/overlay'
        option uuid 'd0f1f31c-68fd-4723-9201-84ecdb2d99a0'
        option enabled '0'

config mount
        option target '/boot'
        option uuid '4DF8-B35F'
        option enabled '0'

config mount
        option target '/rom'
        option uuid '52180f4e-66f311b9-5e10662a-11e42c6c'
        option enabled '0'

config mount
        option target '/overlay'
        option uuid '7d3b4f5e-5dc4-415a-a9f8-7a0bd2cb8891'
        option enabled '1'

root@ImmortalWrt:~# mount
/dev/root on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/loop0 on /overlay type f2fs (rw,lazytime,noatime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,alloc_mode=reuse,fsync_mode=posix)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
/dev/sda1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/sda1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)

```

LawssssCat, 2022-04-20:
With the same configuration, there will be more records after the Lede is restarted, and then the expansion of /overlay is successful

[image]

LawssssCat, 2022-04-20:
How can I choose overlay other than ext4 / f2fs or what can i do to make the expansion effective ?

1715173329, 2022-04-28:
如果你需要扩容，最好的方法是直接使用 ext4 fs 的固件。

closed 2022-05-01 by AmadeusGhost

GreenTeodoro839, 2022-07-02:
> 如果你需要扩容，最好的方法是直接使用 ext4 fs 的固件。

这个问题会修复吗？squashfs更稳定，而且可以系统更新，ext4扩容后更新会无效

LawssssCat, 2022-07-02:
> > 如果你需要扩容，最好的方法是直接使用 ext4 fs 的固件。
> 
> 这个问题会修复吗？squashfs更稳定，而且可以系统更新，ext4扩容后更新会无效

https://github.com/coolsnowwolf/lede/issues/9286#issuecomment-1159376520

这个issue回复提供了思路，但我还没来得及研究

1715173329, 2022-07-02:
Fixed in 90ace9890f07c710c7659142135231a9cf7bc747, which is 62 days ago.

GreenTeodoro839, 2022-07-02:
> Fixed in [90ace98](https://github.com/immortalwrt/immortalwrt/commit/90ace9890f07c710c7659142135231a9cf7bc747), which is 62 days ago.

能不能添加到openwrt-18.06-k5.4分支，该固件https://github.com/SuLingGG/OpenWrt-Rpi使用了这个分支

1715173329, 2022-07-02:
Done.

KardRi, 2024-06-25:
Do you still need this patch if upstream fstools has already fixed it at commit 3a07943, and upstream package fstools Makefile has included since commit c63e3c9?

KardRi, 2024-06-25:
> Done.

Hi, 
Do you still need this patch if upstream fstools has already fixed it at commit 3a07943 (2023-05-29), and upstream package fstools Makefile has included since commit https://github.com/immortalwrt/immortalwrt/commit/c63e3c98e81dff254d4313880d4e231464bb3642 (2023-11-09)?

commit 90ace989 (2022-05-01): fstools: enable any device with non-MTD rootfs_data volume

Fixes: #673
commit da9b81ac (2022-07-02): fstools: enable any device with non-MTD rootfs_data volume

Fixes: #673
(cherry picked from commit 90ace9890f07c710c7659142135231a9cf7bc747)
[rebased patch]
commit 19f355ea (2023-04-13): fstools: enable any device with non-MTD rootfs_data volume

Fixes: #673
(cherry picked from commit 90ace9890f07c710c7659142135231a9cf7bc747)