immortalwrt/immortalwrt #714: AX6S/AX3200 Failed to build

triatomic, 2022-06-13:
Error: ../dts/mt7622-elecom-wrc-x3200gst3.dts:280.1-6 Label or path snfi not found

FATAL ERROR: Syntax error parsing input tree

make[5]: *** [Makefile:31: /home/tria/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/image-mt7622-elecom-wrc-x3200gst3.dtb] Error 1

make[5]: Leaving directory '/home/tria/immortalwrt/target/linux/mediatek/image'

make[4]: *** [Makefile:17: install] Error 2

make[4]: Leaving directory '/home/tria/immortalwrt/target/linux/mediatek'

make[3]: *** [Makefile:11: install] Error 2

make[3]: Leaving directory '/home/tria/immortalwrt/target/linux'

time: target/linux/install#24.70#5.63#30.35

    ERROR: target/linux failed to build.

make[2]: *** [target/Makefile:25: target/linux/install] Error 1

make[2]: Leaving directory '/home/tria/immortalwrt'

make[1]: *** [target/Makefile:19: /home/tria/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2

make[1]: Leaving directory '/home/tria/immortalwrt'

make: *** [/home/tria/immortalwrt/include/toplevel.mk:230: world] Error 2

Compile machine: ubuntu 22.04 LTS on VB up to date.

triatomic, 2022-06-13:
config.buildinfo
```
CONFIG_TARGET_mediatek=y

CONFIG_TARGET_mediatek_mt7622=y

CONFIG_TARGET_mediatek_mt7622_DEVICE_xiaomi_redmi-router-ax6s=y

# CONFIG_PACKAGE_default-settings-chn is not set

CONFIG_PACKAGE_htop=y

CONFIG_PACKAGE_iptables-mod-conntrack-extra=y

CONFIG_PACKAGE_iptables-mod-ipopt=y

CONFIG_PACKAGE_kmod-ifb=y

CONFIG_PACKAGE_kmod-ipt-conntrack-extra=y

CONFIG_PACKAGE_kmod-ipt-ipopt=y

CONFIG_PACKAGE_kmod-nf-conntrack-netlink=y

CONFIG_PACKAGE_kmod-sched-cake=y

CONFIG_PACKAGE_kmod-sched-core=y

CONFIG_PACKAGE_libncurses=y

CONFIG_PACKAGE_librt=y

CONFIG_PACKAGE_libuuid=y

# CONFIG_PACKAGE_luci-app-filetransfer is not set

CONFIG_PACKAGE_luci-app-nlbwmon=y

# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng is not set

# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui is not set

CONFIG_PACKAGE_luci-app-sqm=y

# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_SagerNet_Core is not set

# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Client is not set

# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust_Client is not set

# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust_Server is not set

# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Simple_Obfs is not set

CONFIG_PACKAGE_luci-app-upnp=y

# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_ShadowsocksR_Libev_Server is not set

# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Trojan is not set

# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Xray is not set

# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Xray_plugin is not set

CONFIG_PACKAGE_luci-i18n-nlbwmon-zh-cn=y

CONFIG_PACKAGE_luci-i18n-sqm-zh-cn=y

CONFIG_PACKAGE_luci-i18n-upnp-zh-cn=y

CONFIG_PACKAGE_miniupnpd=y

# CONFIG_PACKAGE_mt7622-preloader is not set

CONFIG_PACKAGE_nlbwmon=y

CONFIG_PACKAGE_sqm-scripts=y

CONFIG_PACKAGE_tc-mod-iptables=y

CONFIG_PACKAGE_tc-tiny=y

CONFIG_PACKAGE_terminfo=y

CONFIG_PACKAGE_zlib=y

# CONFIG_UnblockNeteaseMusic_Go is not set

# CONFIG_UnblockNeteaseMusic_NodeJS is not set

CONFIG_PACKAGE_default-settings=y
```

1715173329, 2022-06-13:
You may simply revert commit `9e6cfcafb6014541865f0402e498f9bb76d407ea` to make it compile for now.

triatomic, 2022-06-13:
> You may simply revert commit `9e6cfcafb6014541865f0402e498f9bb76d407ea` to make it compile for now.
how do i do that ?

1715173329, 2022-06-13:
git revert -n \<commit sha\>

triatomic, 2022-06-13:
Thank you, it has compiled.

closed 2022-06-13 by triatomic

chika0801, 2022-06-14:
刚才测试了，ax6s已经成功编译，3q