immortalwrt/immortalwrt #729: 在MT7628上编译rtl8821cu驱动，编译不过

liuhong785738008, 2022-07-05:
在Openwrt21.02上编译，错误信息如下：
ERROR: "__fixdfsi" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__truncdfsf2" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__floatsidf" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__divdf3" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__adddf3" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__ledf2" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__subdf3" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__muldf3" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__subsf3" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__fixsfsi" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__fixunsdfsi" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__floatunsidf" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__gedf2" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__lesf2" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__ltdf2" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__gtdf2" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!
ERROR: "__floatsisf" [immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/rtl8821cu-2020-12-19-428a0820/rtl8821cu.ko] undefined!

liuhong785738008, 2022-07-05:
大佬们指点一下

1715173329, 2022-07-05:
#726

closed 2022-07-05 by 1715173329

liuhong785738008, 2022-07-05:
--- a/Makefile
+++ b/Makefile
@@ -99,9 +99,9 @@ CONFIG_AP_WOWLAN = n
 ######### Notify SDIO Host Keep Power During Syspend ##########
 CONFIG_RTW_SDIO_PM_KEEP_POWER = y 
 ###################### MP HW TX MODE FOR VHT #######################
-CONFIG_MP_VHT_HW_TX_MODE = y
+CONFIG_MP_VHT_HW_TX_MODE = n
 ###################### Platform Related #######################
-CONFIG_PLATFORM_I386_PC = y
+CONFIG_PLATFORM_I386_PC = n
 CONFIG_PLATFORM_ARM_RPI = n 
 CONFIG_PLATFORM_ARM64_RPI = n 
 CONFIG_PLATFORM_ANDROID_X86 = n 
@@ -113,6 +113,7 @@ CONFIG_PLATFORM_ARM_S3C6K4 = n
 CONFIG_PLATFORM_MIPS_RMI = n 
 CONFIG_PLATFORM_RTD2880B = n 
 CONFIG_PLATFORM_MIPS_AR9132 = n 
+CONFIG_PLATFORM_MIPS_MT7628 = y
 CONFIG_PLATFORM_RTK_DMP = n 
 CONFIG_PLATFORM_MIPS_PLM = n 
 CONFIG_PLATFORM_MSTAR389 = n 
@@ -1443,6 +1444,18 @@ CROSS_COMPILE := mips-openwrt-linux-
 KSRC := /home/alex/test_openwrt/tmp/linux-2.6.30.9
 endif
 
+ifeq ($(CONFIG_PLATFORM_MIPS_MT7628), y)
+ifeq ($(CONFIG_CONCURRENT_MODE), y)
+EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
+endif
+EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
+EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
+ARCH := mips 
+CROSS_COMPILE ?=
+KSRC := $(TOPDIR)/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-5.4.188
+endif
+
 ifeq ($(CONFIG_PLATFORM_DMP_PHILIPS), y)
 EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN -DRTK_DMP_PLATFORM
 ARCH := mips

上面是我打的补丁，已经可以编译通过，感谢