immortalwrt/immortalwrt #737: 最近有没有人出现tproxy不好使的情况？

dfldylan, 2022-07-17:
换成redirect就好使，大约从7.13开始编译的固件都有这个问题。
不是到哪个包的问题。。。排查好几天没找出来。
用官方固件编译也有这个问题

1715173329, 2022-07-17:
snapshots 自求多福。

zxlhhyccc, 2022-07-17:
> 换成redirect就好使，大约从7.13开始编译的固件都有这个问题。 不是到哪个包的问题。。。排查好几天没找出来。 用官方固件编译也有这个问题

请说明具体的表现，貌似我没问题。。。。

dfldylan, 2022-07-17:
我在R4S和x86上都复现了这个问题，openwrt22和21版本都尝试过。我主要使用https://github.com/kuoruan/luci-app-v2ray或者https://github.com/yichya/luci-app-xray。具体表现为，不用tproxy用redirect时就正常代理（用NAT表）：
[image]
[image]
一旦换成tproxy模式（用mangle表，为了方便对比不代理UDP）：
[image]
就代理不上去了。

第一次发现这个问题大约是13号开始编译的固件，之后一直都是这样。
更奇怪的是我尝试固定feeds.conf.default中的git历史版本，还是会有这样的情况。

dfldylan, 2022-07-17:
我只保留了luci-app-xray，这个项目使用tproxy。其他的什么都没有选。由于这个项目默认是没有ipv6的，因此在只有ipv4的环境下，默认配置有几率能连上外网，但是很慢。后期我又加了ipv6helper又编译了一下，这回一点都上不去了。。我在想是不是最近哪个kmod有更新

dfldylan, 2022-07-23:
我又用GL-INET的AXT1800试了一下，这个是拉的是官方21.02的openwrt。同样的配置在7.13之前编译的固件就能透明代理，今天编译的固件就不行

closed 2022-07-23 by dfldylan