immortalwrt/immortalwrt #740: rtl88x2bu 无线网络重启后无法ping通

qbisi, 2022-07-25:
系统：openwrt-21.02 aarch64(rockchip)
无线网卡：cf-812ac (rtl88x2bu驱动)
无线连接模式：openwrt routed ap mode (https://openwrt.org/docs/guide-user/network/wifi/routedap)

bug现象： 系统启动后能ping通wlan口网关（192.168.1.1），ifup wlan 或者wifi up radio1 后有无线链路但无法ping通wlan口网关

bug排除：关闭防火墙和dhcp, 接入端设置静态ip 192.168.2.100,仍然无法ping通网关

临时解决方案： rmmod rtl88x2bu && modprobe rtl88x2bu && ifup wlan

配置文件：
/etc/config/network:

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fde4:e33a:6e84::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '172.16.1.1'
	option netmask '255.255.254.0'

config interface 'wlan'
	option proto 'static'
	option device 'wlan0'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
```

/etc/config/wireless
```
config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/fe380000.usb/usb1/1-1/1-1.2/1-1.2:1.0'
        option band '5g'
        option rx_stbc '0'   #must be 0 here or hostapd won't start the second time 
        option cell_density '0'
        option htmode 'VHT80'
        option channel '36'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option network 'wlan'
```

1715173329, 2022-07-25:
RTL 网卡驱动问题很多，影响因素也很多，建议买 mt76x2u 芯片的网卡。

brctl show br-lan 看下启动的时候有没有 wlan。

qbisi, 2022-07-25:
> 

我没有把wlan口桥接到br-lan上，wlan和lan是分开的两个网段。开机后的无线网络是正常的，ifup wlan 后重新加载网络会有问题，所以我感觉问题不在驱动。

1715173329, 2022-07-25:
rtl 驱动随便做点什么都得重启的，需要 rmmod 也在情理之中。
但既然能用又为什么要去重载？

qbisi, 2022-07-25:
我想定位下rtl网卡的问题，因为目前master分支用rtl8821cu和rtl88x2bu都会有问题（我的板子是反复重启）。可能对于解决master分支有帮助。

qbisi, 2022-07-25:
另外我在使用rtl8821cu的时候，板子第一次上电（非重启）不能正常配置无线网络，这样随身带（只用无线管理）就没法用了
所以我在自定义启动项里面加入了
rmmod rtl8821cu && modprobe rtl8821cu && service network restart
但这只是临时解决方案.

1715173329, 2022-07-25:
https://github.com/immortalwrt/immortalwrt/blob/a4669fa7c3ebf10dd82c49ddcf8ff7811abc8d44/package/kernel/rtl8821cu/Makefile#L36
可以试试更改一下加载顺序，比如 `AUTOLOAD:=$(call AutoProbe,99,rtl8821cu)`

qbisi, 2022-07-25:
> https://github.com/immortalwrt/immortalwrt/blob/a4669fa7c3ebf10dd82c49ddcf8ff7811abc8d44/package/kernel/rtl8821cu/Makefile#L36
> 
> 
> 可以试试更改一下加载顺序，比如 `AUTOLOAD:=$(call AutoProbe,99,rtl8821cu)`

似乎不行，这样改开机不会加载rtl8821cu模块

qbisi, 2022-07-26:
更新一下我目前的解决方案
修改/sbin/wifi 
```
diff /sbin/wifi /sbin/wifi.old

wifi_updown() {
        cmd=down
        [ enable = "$1" ] && {
                _wifi_updown disable "$2"
                ubus_wifi_cmd "$cmd" "$2"

+                rmmod rtl88x2bu

                scan_wifi
                cmd=up
                ubus call network reload
        }
        [ reconf = "$1" ] && {
                scan_wifi
                cmd=reconf
                ubus call network reload
        }

+        modprobe rtl88x2bu

        ubus_wifi_cmd "$cmd" "$2"
        _wifi_updown "$@"
}
```
不规范的来说，能否将rtl88x2 的 modprobe 行为关联到 ubus_wifi_cmd network up | down 行为中，只修改和rtl88x2有关的代码解决这一问题。

qbisi, 2022-07-27:
7.28测试
1. rmmod rtl88x2bu
2. modprobe rtl88x2bu 后会产生两个网口wlan0,wlan1 
3. ubus call network.wireless up 这两个网口才会变成一个，同时会生成/var/run/hostpad-phy*.conf 配置文件，
但是生成的hostpad-phy*.conf 中 vht_capab没有把驱动提供的功能用上
4. 再次执行 ubus call network.wireless up ,生成的hostpad-phy*.conf 中 vht_capab是完整的，但是这时候会出现无法ping通的现象。

我的问题：如何查看ubus call network.wireless up这一注册函数，怀疑这里面加了正确的modprobe option。

qbisi, 2024-03-05:
嗯，这个issue可以取消了。