immortalwrt/immortalwrt #759: 21.02x86编译失败

jh0072110, 2022-08-17:
以下为日志文件错误部分节选以及全部：
make[3]: Entering directory '/workdir/openwrt/package/kernel/r8101'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.r8101-1.037.01.tar.bz2.flock -c '  	echo "ERROR: No download method available"; false'
ERROR: No download method available
make[3]: *** [Makefile:40: /workdir/openwrt/dl/r8101-1.037.01.tar.bz2] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/r8101'
time: package/kernel/r8101/compile#1.24#0.12#1.36
    ERROR: package/kernel/r8101 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/r8101/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.
[11_Compile the firmware.txt](https://github.com/immortalwrt/immortalwrt/files/9356241/11_Compile.the.firmware.txt)

umarizulkifli, 2022-08-17:
https://github.com/immortalwrt/immortalwrt/issues/758
No package source url https://github.com/immortalwrt/immortalwrt/blob/af6bba0dd7534fe4303f6d81c4a4b7c951b33522/package/kernel/r8101/Makefile#L13

https://github.com/umarizulkifli/r8101/raw/main/r8101-1.037.01.tar.bz2

umarizulkifli, 2022-08-17:
PR submitted
https://github.com/immortalwrt/immortalwrt/pull/760

1715173329, 2022-08-17:
Fixed in f150238b1d905a547e017e12eaa1605b913a4265.

commit 8c8ba72a (2022-08-17): download.pl: add ImmortalWrt fallback

Fixes: #759

(cherry picked from commit a19080494926ad523e636d7e075717b69e2f3e47)
closed 2022-08-17 by 1715173329