immortalwrt/immortalwrt #762: X710-T4L不支持2.5G和5G

shenlijun, 2022-08-24:
RT，按理说不应该啊

nicholas-opensource, 2022-08-24:
您好，您的问题已经收到。您可以先尝试用ethtool先查看网卡的输出信息来帮助判断。如果您目前使用的是k5.4内核，那么有可能是i40e驱动太过于老旧，需要backport到最新的驱动。

shenlijun, 2022-08-25:
> 您好，您的问题已经收到。您可以先尝试用ethtool先查看网卡的输出信息来帮助判断。如果您目前使用的是k5.4内核，那么有可能是i40e驱动太过于老旧，需要backport到最新的驱动。

我编译的是master分支，5.10内核。我发现Ubuntu22.04上也一样没有2.5G和5G，不知道是不是驱动本身的问题
[image]

还有发现一些在i40e上关于2.5G之类的patch。不知道X710-T4L是不是也需要专门改一下才行
[image]

shenlijun, 2022-08-25:
还有X550也很有意思，Supported link mode有2.5G和5G，但是Advertised link mode却没有

nicholas-opensource, 2022-08-25:
您好。您可以试着使用最新的内核（5.19）来检查一下一切是否正常。我在k5.10的i40e源码中的确发现了2_5G BaseT和5G BaseT的字样。此外我很确定T2L在Windows server下安装intel官网驱动包mGig是一切正常的。
此外，在[这篇文章](https://www.servethehome.com/intel-x710-t4-quad-port-10gbase-t-nic-review/)中，的确提到了可以正确使用mGig。
如果您依然存在问题，建议可以去intel社区进行提问。

shenlijun, 2022-08-25:
NTEL很迷啊，我刚才又特意试了下X550，官方明确说windows不支持2.5G，但是刚才自动协商出了2.5G。然后我安装了个官网最新的驱动，瞬间变成了1G😄

nicholas-opensource, 2022-08-25:
所以我建议这件事可以在intel官方论坛发帖讨论，这怎么看都是属于他们的问题

shenlijun, 2022-08-25:
> 您好。您可以试着使用最新的内核（5.19）来检查一下一切是否正常。我在k5.10的i40e源码中的确发现了2_5G BaseT和5G BaseT的字样。此外我很确定T2L在Windows server下安装intel官网驱动包mGig是一切正常的。 此外，在[这篇文章](https://www.servethehome.com/intel-x710-t4-quad-port-10gbase-t-nic-review/)中，的确提到了可以正确使用mGig。 如果您依然存在问题，建议可以去intel社区进行提问。

https://www.bilibili.com/read/cv6319233/
这里下面的第二条评论说T2L也会跳过2.5G

结合刚才x550的现象来看，可能还是官方驱动更新的过程中有什么改动

nicholas-opensource, 2022-08-25:
所以解决办法就是
1）联系intel官方技术专员或者您的供货商，以及在intel论坛提问。
2）现实中的实践经验表明，2.5G 5G的mGig带来的巨量的不兼容和不稳定问题，目前还不是很推荐使用（如果您用于生产环境）

shenlijun, 2022-08-25:
> 所以解决办法就是 1）联系intel官方技术专员或者您的供货商，以及在intel论坛提问。 2）现实中的实践经验表明，2.5G 5G的mGig带来的巨量的不兼容和不稳定问题，目前还不是很推荐使用（如果您用于生产环境）

悲剧，看来只能等官方更新了
https://sourceforge.net/p/e1000/mailman/message/37106343/

closed 2022-08-25 by shenlijun