immortalwrt/immortalwrt #787: 官网下载的固件没有识别2.5G网卡 R8125

boyu899, 2022-10-09:
immortalwrt官网下载的固件没有识别2.5G网卡 R8125  软路由J4125  4口2.5G网卡   lede固件可以 其它集合插件或纯净版(lede源码)的也有  自己编译了lede源码 识别8125 基本不用设置 直接编译    天灵源码  不行哦  使用https://github.com/SuLingGG/OpenWrt-Rpi固件 貌似也是[immortalwrt]源码  也只能识别到1000M

boyu899, 2022-10-09:
官网下载的x84固件自带r8125 就是识别不了 看了下软件源 已安装了驱动

1715173329, 2022-10-09:
那么你换一个你认为能用的固件即可。

imoldchen, 2022-10-26:
识别 1000M 好像去年某个更新驱动版本之后就这样了

1715173329, 2022-10-26:
https://github.com/awesometic/realtek-r8125-dkms/blob/master/REALTEK_README.txt

You may enforce 2.5 Gpbs via cli.

xchgeaxeax, 2023-01-08:
碰到同样的问题了，有个方案可以参考一下，我现在临时解决方案是修改/etc/rc.local，把eth*换成你实际网卡
`# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

# r8125 2500mbps autoneg
ethtool -s eth2 autoneg on advertise 0x80000000002f
ethtool -s eth3 autoneg on advertise 0x80000000002f
ethtool -s eth4 autoneg on advertise 0x80000000002f
ethtool -s eth5 autoneg on advertise 0x80000000002f

exit 0
`

codgician, 2023-01-21:
Another solution is to add a `hotplug` script to enforce speed when `br-lan` is brought up:

An example for bring up `eth1`: create `/etc/hotplug.d/iface/01-fix-autoneg` with:

```bash
# Fix for RTL8125 speed auto negotiation
[ "ifup" == "$ACTION" ] && [ "br-lan" == "$DEVICE" ] {
        logger -t fix-autoneg "Fix auto negotiation for eth1 as $DEVICE has action $ACTION"
        ethtool -s eth1 autoneg on advertise 0x80000000002f
}
```

1715173329, 2023-01-27:
https://github.com/x-wrt/x-wrt/tree/d3c5947/target/linux/generic/backport-5.4
加上这里的补丁试试呢？不行的话大概可能就需要做个 luci 之类的玩意儿了？

sbwml, 2023-01-30:
和我最开始 nanopi r5s 用openwrt 一个情况，驱动的问题。简单直接的方法是直接用 r8169 ，用最新的 r8125 你可能需要改一下下源码。

1715173329, 2023-02-24:
Try if #885 works for you

1715173329, 2023-02-24:
fixed by 891bf5b61db9901c5474850bc644e4730b981037

closed 2023-02-24 by 1715173329