immortalwrt/immortalwrt #803: Rockchip: master分支编译失败(无限循环)

xlighting2017, 2022-10-24:
日志如下（重复出现的行 我删掉了，再往下就是 无限循环）：

```bash
2022-10-23T15:29:01.1797289Z # configuration written to .config
2022-10-23T15:29:01.1797562Z #
2022-10-23T15:29:01.1818323Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10'
2022-10-23T15:29:01.1819960Z /workdir/openwrt/staging_dir/host/bin/sed -i -e 's#CONFIG_MKIMAGE_DTC_PATH=.*#CONFIG_MKIMAGE_DTC_PATH="/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10/scripts/dtc/dtc"#g' /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10/.config
2022-10-23T15:29:01.1855963Z touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10/.configured_68b329da9893e34099c7d8ad5cb9c940
2022-10-23T15:29:01.1883089Z rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10/.built
2022-10-23T15:29:01.1915426Z touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10/.built_check
2022-10-23T15:29:01.1944019Z make  -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10 CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.74/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib -L/workdir/openwrt/staging_dir/hostpkg/lib -L/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-ImmortalWrt-r24087-996ef9f206" STAGING_PREFIX="/workdir/openwrt/staging_dir/host" PKG_CONFIG_PATH="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' BL31=/workdir/openwrt/staging_dir/target-aarch64_generic_musl/image/rk322xh_bl31_v1.49.elf
2022-10-23T15:29:01.1978425Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2c-rk3328/u-boot-2022.10'
2022-10-23T15:29:01.5647345Z   HOSTCC  scripts/basic/fixdep
2022-10-23T15:29:01.7323057Z   HOSTCC  scripts/kconfig/conf.o
2022-10-23T15:29:01.9404218Z   HOSTCC  scripts/kconfig/zconf.tab.o
2022-10-23T15:29:03.9483716Z   HOSTLD  scripts/kconfig/conf
2022-10-23T15:29:03.9783741Z scripts/kconfig/conf  --syncconfig Kconfig
2022-10-23T15:29:04.1982733Z .config:520:warning: symbol value '' invalid for SPL_BSS_MAX_SIZE
2022-10-23T15:29:04.1998839Z .config:1849:warning: symbol value '' invalid for SYS_USB_OHCI_MAX_ROOT_PORTS
2022-10-23T15:29:04.2079034Z 
2022-10-23T15:29:04.2079326Z *
2022-10-23T15:29:04.2081302Z * Restart config...
2022-10-23T15:29:04.2081759Z *
2022-10-23T15:29:04.2083682Z Error in reading or end of file.
2022-10-23T15:29:04.2083957Z *
2022-10-23T15:29:04.2084505Z * SPL configuration options
2022-10-23T15:29:04.2084765Z *
2022-10-23T15:29:04.2085016Z 
2022-10-23T15:29:04.2085149Z Error in reading or end of file.
2022-10-23T15:29:04.2085333Z 
2022-10-23T15:29:04.2085442Z Error in reading or end of file.
2022-10-23T15:29:04.2085638Z 
2022-10-23T15:29:04.2085758Z Error in reading or end of file.
2022-10-23T15:29:04.2085936Z 
2022-10-23T15:29:04.2086059Z Error in reading or end of file.
2022-10-23T15:29:04.2086240Z 
2022-10-23T15:29:04.2086362Z Error in reading or end of file.
2022-10-23T15:29:04.2086539Z 
2022-10-23T15:29:04.2086661Z Error in reading or end of file.
2022-10-23T15:29:04.2086818Z 
2022-10-23T15:29:04.2086938Z Error in reading or end of file.
2022-10-23T15:29:04.2087464Z 
2022-10-23T15:29:04.2087589Z Error in reading or end of file.
....
2022-10-23T15:29:04.3117218Z Error in reading or end of file.
2022-10-23T15:29:04.3117225Z 
2022-10-23T15:29:04.3117328Z Error in reading or end of file.
2022-10-23T15:29:04.3117335Z 
2022-10-23T15:29:04.3117438Z Error in reading or end of file.
2022-10-23T15:29:04.3117444Z 
2022-10-23T15:29:04.3117547Z Error in reading or end of file.
2022-10-23T15:29:04.3117762Z Support SPL based upon the common SPL framework (SPL_FRAMEWORK) [Y/n/?] y
2022-10-23T15:29:04.3117949Z   Define a generic function board_init_f (SPL_FRAMEWORK_BOARD_INIT_F) [N/y/?] n
2022-10-23T15:29:04.3118098Z Maximum size of SPL image (SPL_SIZE_LIMIT) [0x0] 0x0
2022-10-23T15:29:04.3118269Z Maximum size of the SPL image, excluding BSS (SPL_MAX_SIZE) [0x0] 0x0
2022-10-23T15:29:04.3118504Z Offset to which the SPL should be padded before appending the SPL payload (SPL_PAD_TO) [0x7000] 0x7000
2022-10-23T15:29:04.3118723Z Use a specific address for the BSS via the linker script (SPL_HAS_BSS_LINKER_SECTION) [N/y/?] n
2022-10-23T15:29:04.3118822Z Enforce SPL BSS limit
2022-10-23T15:29:04.3119012Z   1. Do not enforce a build time limit on the size of the BSS (SPL_NO_BSS_LIMIT)
2022-10-23T15:29:04.3119172Z > 2. Enforce a limit on the size of the BSS only (SPL_BSS_LIMIT)
2022-10-23T15:29:04.3119384Z   3. Enforce a limit on the whole of memory allocated to SPL, BSS included (SPL_FOOTPRINT_LIMIT)
2022-10-23T15:29:04.3119744Z choice[1-3?]: 2
2022-10-23T15:29:04.3119930Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
2022-10-23T15:29:04.3120103Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
2022-10-23T15:29:04.3120280Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
2022-10-23T15:29:04.3120459Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
...
2022-10-23T15:29:04.3997352Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
2022-10-23T15:29:04.3997536Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
2022-10-23T15:29:04.3997724Z Maximum size in memory allocated to the SPL BSS (SPL_BSS_MAX_SIZE) [] (NEW) 
2022-10-23T15:29:04.4018493Z 
2022-10-23T15:29:04.4018673Z Error in reading or end of file.
2022-10-23T15:29:04.4018681Z 
2022-10-23T15:29:04.4018804Z Error in reading or end of file.
2022-10-23T15:29:04.4018811Z 
2022-10-23T15:29:04.4018928Z Error in reading or end of file.
2022-10-23T15:29:04.4018935Z 
2022-10-23T15:29:04.4019053Z Error in reading or end of file
....
2022-10-23T15:29:04.4510805Z 
2022-10-23T15:29:04.4510916Z Error in reading or end of file.
2022-10-23T15:29:04.4510922Z 
2022-10-23T15:29:04.4511036Z Error in reading or end of file.
2022-10-23T15:29:04.4511045Z 
2022-10-23T15:29:04.4511159Z Error in reading or end of file.
```

1715173329, 2022-10-24:
Do not use verbose build.

xlighting2017, 2022-10-24:
it has nothing to do with verbose,  below is the log when using "make j=1" (without V=s)

[image]

it indicates that the "uboot-rockchip" is compiling forever

1715173329, 2022-10-24:
fixed.

commit 9360a9e0 (2022-10-24): uboot-rockchip: refresh defconfigs

Fixes: c88a74b6350b3d ("uboot-rockchip: update to v2022.10")
Fixes: #803
closed 2022-10-24 by 1715173329