immortalwrt/immortalwrt #805: RTL8821CU/RTL8811CU not working

digik-git, 2022-10-30:
The RTL USB wifi with Realtek RTL8811CU/8821CU does not work despite i've installed the driver kmod-rtl8821cu_5.4.215+2022-04-09-8c2226a7-1_arm_arm1176jzf-s_vfp.ipk for a Raspberry PI.

It shows only "generic 802.11bg device is not active"

wittcong, 2022-11-02:
You can try adding USB2.0&3.0 drivers.

nonenty, 2023-05-17:
I'm facing the same problem. The USB wifi device I'm using is TL-WDN5200H. Its USB id is 0bda:c811, so its chipset is RTL8821CU too according to [here](https://linux-hardware.org/index.php?id=usb:0bda-c811).
I installed the immortalwrt in a virtual machine on pve using the image file downloaded from [here](https://downloads.immortalwrt.org/releases/21.02.6/targets/x86/64/immortalwrt-21.02.6-x86-64-generic-ext4-combined-efi.img.gz). I then installed kmod-rtl8821cu and all the kmod-usb* packages but it still failed to work.

here is part of the kernal log:
```plain
[    6.475256] rtl8821cu: Unknown symbol __cfg80211_alloc_event_skb (err -2)
[    6.476350] rtl8821cu: Unknown symbol wiphy_register (err -2)
[    6.477262] rtl8821cu: Unknown symbol cfg80211_remain_on_channel_expired (err -2)
[    6.478377] rtl8821cu: Unknown symbol cfg80211_vendor_cmd_reply (err -2)
[    6.479372] rtl8821cu: Unknown symbol cfg80211_del_sta_sinfo (err -2)
[    6.480370] rtl8821cu: Unknown symbol wiphy_unregister (err -2)
[    6.481285] rtl8821cu: Unknown symbol cfg80211_ch_switch_notify (err -2)
[    6.482320] rtl8821cu: Unknown symbol ieee80211_get_channel_khz (err -2)
[    6.483304] rtl8821cu: Unknown symbol cfg80211_ready_on_channel (err -2)
[    6.484355] rtl8821cu: Unknown symbol wiphy_free (err -2)
[    6.485244] rtl8821cu: Unknown symbol wiphy_new_nm (err -2)
[    6.486172] rtl8821cu: Unknown symbol cfg80211_get_bss (err -2)
[    6.487083] rtl8821cu: Unknown symbol ieee80211_freq_khz_to_channel (err -2)
[    6.488097] rtl8821cu: Unknown symbol __cfg80211_send_event_skb (err -2)
[    6.489094] rtl8821cu: Unknown symbol cfg80211_disconnected (err -2)
[    6.490086] rtl8821cu: Unknown symbol cfg80211_michael_mic_failure (err -2)
[    6.491094] rtl8821cu: Unknown symbol cfg80211_ibss_joined (err -2)
[    6.492100] rtl8821cu: Unknown symbol cfg80211_scan_done (err -2)
[    6.493063] rtl8821cu: Unknown symbol cfg80211_roamed (err -2)
[    6.494008] rtl8821cu: Unknown symbol cfg80211_put_bss (err -2)
[    6.494937] rtl8821cu: Unknown symbol cfg80211_connect_done (err -2)
[    6.495936] rtl8821cu: Unknown symbol cfg80211_unlink_bss (err -2)
[    6.496856] rtl8821cu: Unknown symbol cfg80211_inform_bss_frame_data (err -2)
[    6.497872] rtl8821cu: Unknown symbol cfg80211_new_sta (err -2)
[    6.498807] rtl8821cu: Unknown symbol __cfg80211_alloc_reply_skb (err -2)
[    6.499822] rtl8821cu: Unknown symbol wiphy_apply_custom_regulatory (err -2)
[    6.500808] rtl8821cu: Unknown symbol cfg80211_mgmt_tx_status (err -2)
[    6.501803] rtl8821cu: Unknown symbol cfg80211_rx_mgmt_khz (err -2)
[    6.515005] kmodloader: 1 module could not be probed
[    6.516838] kmodloader: - rtl8821cu - 0
```
I tried
```
cat /proc/kallsyms | grep cfg80211
```
but there's no output. 

It seems that there's something wrong with the kmod-cfg80211, but it has already been compiled into the firmware. Any solutions?

1715173329, 2023-05-17:
no idea what's wrong with it, rtl drivers are shitty and buggy.

try this driver and see if it works for you
https://github.com/1715173329/imoutowrt/tree/2102-rtw88

1715173329, 2024-01-06:
Upstream has added support for this chip, use `kmod-rtw88-8821cu` instead.

closed 2024-01-06 by 1715173329