immortalwrt/immortalwrt #820: 大神,能否支持电犀牛r66s(fastrhino_r66s)

yanght1001, 2022-11-17:
感觉大神的这个库真心好用,最近购入了一个R66S,发现不支持,请求支持,感谢大神,,rockchip-armv8-fastrhino_r66s

closed 2022-11-17 by 1715173329

Ocean-TXT, 2024-09-27:
请问现在支持R66S了吗？

rootphantomer, 2024-09-27:
> 请问现在支持R66S了吗？

https://firmware-selector.immortalwrt.org/?version=23.05.3&target=rockchip%2Farmv8&id=lunzn_fastrhino-r66s

Ocean-TXT, 2024-09-27:
> > 请问现在支持R66S了吗？
> 
> https://firmware-selector.immortalwrt.org/?version=23.05.3&target=rockchip%2Farmv8&id=lunzn_fastrhino-r66s

 太干净了，服务都没有 哭  怎么安装openclash喃

ksong008, 2024-09-27:
@Ocean-TXT 在线直接装……

Ocean-TXT, 2024-09-27:
> 在线直接装......

ssh 链接不上，软件包安装了提示 Package luci-app-openclash (0.46.033-beta) installed in root is up to date.

Ocean-TXT, 2024-09-27:
SSH 连上马上断开 openclash软件包安装后没有服务

rootphantomer, 2024-09-27:
> SSH 连上马上断开 openclash软件包安装后没有服务

人家这边是做纯净系统的，你软件该咋装就咋装。就跟你去问微软，qq 咋装。你要封装好的就去其他集成 os找固件。

1715173329, 2024-09-27:
自己用隐身模式打开就行了。这里是 issue，不是讨论区。

libiunc, 2024-09-27:
使用immortalwrt selector  请求构建包里面带上`luci-app-openclash`，一个漏油器系统不至于做这么离大普的隐私防护