immortalwrt/immortalwrt #834: 拨号偶尔会断

Karmylr, 2022-12-11:
RT，拨号偶尔会断一个小时。用小米硬路由拨号就没这个问题。网络-接口处显示未知错误。以下是日志：
Sat Dec 10 23:56:33 2022 daemon.warn pppd[6979]: Timeout waiting for PADO packets
Sat Dec 10 23:56:33 2022 daemon.err pppd[6979]: Unable to complete PPPoE Discovery
Sat Dec 10 23:56:33 2022 daemon.info pppd[6979]: Exit.
Sat Dec 10 23:56:33 2022 daemon.notice netifd: Interface 'WAN' is now down
Sat Dec 10 23:56:33 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sat Dec 10 23:56:33 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sat Dec 10 23:56:33 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sat Dec 10 23:56:33 2022 kern.info kernel: [  135.248696] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat Dec 10 23:56:33 2022 kern.info kernel: [  135.249983] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Sat Dec 10 23:56:33 2022 kern.info kernel: [  135.250579] 8021q: adding VLAN 0 to HW filter on device eth1
Sat Dec 10 23:56:33 2022 daemon.err insmod: module is already loaded - slhc
Sat Dec 10 23:56:33 2022 daemon.err insmod: module is already loaded - ppp_generic
Sat Dec 10 23:56:33 2022 daemon.err insmod: module is already loaded - pppox
Sat Dec 10 23:56:33 2022 daemon.err insmod: module is already loaded - pppoe
Sat Dec 10 23:56:33 2022 daemon.notice netifd: WAN (7087): uci: Entry not found
Sat Dec 10 23:56:33 2022 daemon.notice netifd: WAN (7087): sh: out of range
Sat Dec 10 23:56:33 2022 daemon.info pppd[7097]: Plugin rp-pppoe.so loaded.
Sat Dec 10 23:56:33 2022 daemon.info pppd[7097]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat Dec 10 23:56:33 2022 daemon.notice pppd[7097]: pppd 2.4.8 started by root, uid 0
Sat Dec 10 23:56:48 2022 daemon.warn pppd[7097]: Timeout waiting for PADO packets
Sat Dec 10 23:56:48 2022 daemon.err pppd[7097]: Unable to complete PPPoE Discovery
Sat Dec 10 23:56:48 2022 daemon.info pppd[7097]: Exit.
Sat Dec 10 23:56:48 2022 daemon.notice netifd: Interface 'WAN' is now down
Sat Dec 10 23:56:48 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sat Dec 10 23:56:48 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sat Dec 10 23:56:48 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sat Dec 10 23:56:48 2022 kern.info kernel: [  150.384954] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat Dec 10 23:56:48 2022 kern.info kernel: [  150.386106] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Sat Dec 10 23:56:48 2022 kern.info kernel: [  150.386679] 8021q: adding VLAN 0 to HW filter on device eth1
Sat Dec 10 23:56:48 2022 daemon.err insmod: module is already loaded - slhc
Sat Dec 10 23:56:48 2022 daemon.err insmod: module is already loaded - ppp_generic
Sat Dec 10 23:56:48 2022 daemon.err insmod: module is already loaded - pppox
Sat Dec 10 23:56:48 2022 daemon.err insmod: module is already loaded - pppoe
Sat Dec 10 23:56:48 2022 daemon.notice netifd: WAN (7317): uci: Entry not found
Sat Dec 10 23:56:48 2022 daemon.notice netifd: WAN (7317): sh: out of range
Sat Dec 10 23:56:48 2022 daemon.info pppd[7327]: Plugin rp-pppoe.so loaded.
Sat Dec 10 23:56:48 2022 daemon.info pppd[7327]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat Dec 10 23:56:49 2022 daemon.notice pppd[7327]: pppd 2.4.8 started by root, uid 0
Sat Dec 10 23:57:09 2022 daemon.info pppd[7327]: PPP session is 34497
Sat Dec 10 23:57:09 2022 daemon.warn pppd[7327]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
Sat Dec 10 23:57:09 2022 kern.info kernel: [  170.487901] pppoe-WAN: renamed from ppp0
Sat Dec 10 23:57:09 2022 daemon.info pppd[7327]: Renamed interface ppp0 to pppoe-WAN
Sat Dec 10 23:57:09 2022 daemon.info pppd[7327]: Using interface pppoe-WAN
Sat Dec 10 23:57:09 2022 daemon.notice pppd[7327]: Connect: pppoe-WAN <--> eth1
Sat Dec 10 23:57:12 2022 daemon.info pppd[7327]: syncppp not active
Sat Dec 10 23:57:39 2022 daemon.notice pppd[7327]: Modem hangup
Sat Dec 10 23:57:39 2022 daemon.notice pppd[7327]: Connection terminated.
Sat Dec 10 23:57:39 2022 daemon.info pppd[7327]: Sent PADT
Sat Dec 10 23:57:39 2022 daemon.info pppd[7327]: Exit.
Sat Dec 10 23:57:39 2022 daemon.notice netifd: Interface 'WAN' is now down
Sat Dec 10 23:57:39 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sat Dec 10 23:57:39 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sat Dec 10 23:57:39 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sat Dec 10 23:57:39 2022 kern.info kernel: [  200.526290] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat Dec 10 23:57:39 2022 kern.info kernel: [  200.527462] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Sat Dec 10 23:57:39 2022 kern.info kernel: [  200.528050] 8021q: adding VLAN 0 to HW filter on device eth1
Sat Dec 10 23:57:39 2022 daemon.err insmod: module is already loaded - slhc
Sat Dec 10 23:57:39 2022 daemon.err insmod: module is already loaded - ppp_generic
Sat Dec 10 23:57:39 2022 daemon.err insmod: module is already loaded - pppox
Sat Dec 10 23:57:39 2022 daemon.err insmod: module is already loaded - pppoe
Sat Dec 10 23:57:39 2022 daemon.notice netifd: WAN (7694): uci: Entry not found
Sat Dec 10 23:57:39 2022 daemon.notice netifd: WAN (7694): sh: out of range
Sat Dec 10 23:57:39 2022 daemon.info pppd[7704]: Plugin rp-pppoe.so loaded.
Sat Dec 10 23:57:39 2022 daemon.info pppd[7704]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat Dec 10 23:57:39 2022 daemon.notice pppd[7704]: pppd 2.4.8 started by root, uid 0
Sat Dec 10 23:57:54 2022 daemon.warn pppd[7704]: Timeout waiting for PADO packets
Sat Dec 10 23:57:54 2022 daemon.err pppd[7704]: Unable to complete PPPoE Discovery
Sat Dec 10 23:57:54 2022 daemon.info pppd[7704]: Exit.
Sat Dec 10 23:57:54 2022 daemon.notice netifd: Interface 'WAN' is now down
Sat Dec 10 23:57:54 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sat Dec 10 23:57:54 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sat Dec 10 23:57:54 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sat Dec 10 23:57:54 2022 kern.info kernel: [  215.644839] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat Dec 10 23:57:54 2022 kern.info kernel: [  215.645977] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Sat Dec 10 23:57:54 2022 kern.info kernel: [  215.646557] 8021q: adding VLAN 0 to HW filter on device eth1
Sat Dec 10 23:57:54 2022 daemon.err insmod: module is already loaded - slhc
Sat Dec 10 23:57:54 2022 daemon.err insmod: module is already loaded - ppp_generic
Sat Dec 10 23:57:54 2022 daemon.err insmod: module is already loaded - pppox
Sat Dec 10 23:57:54 2022 daemon.err insmod: module is already loaded - pppoe
Sat Dec 10 23:57:54 2022 daemon.notice netifd: WAN (7808): uci: Entry not found
Sat Dec 10 23:57:54 2022 daemon.notice netifd: WAN (7808): sh: out of range
Sat Dec 10 23:57:54 2022 daemon.info pppd[7818]: Plugin rp-pppoe.so loaded.
Sat Dec 10 23:57:54 2022 daemon.info pppd[7818]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat Dec 10 23:57:54 2022 daemon.notice pppd[7818]: pppd 2.4.8 started by root, uid 0
Sat Dec 10 23:58:14 2022 daemon.err pppd[7818]: select (waitForPADS): Interrupted system call
Sat Dec 10 23:58:14 2022 daemon.warn pppd[7818]: Timeout waiting for PADS packets
Sat Dec 10 23:58:14 2022 daemon.err pppd[7818]: Unable to complete PPPoE Discovery
Sat Dec 10 23:58:14 2022 daemon.info pppd[7818]: Exit.
Sat Dec 10 23:58:14 2022 daemon.notice netifd: Interface 'WAN' is now down
Sat Dec 10 23:58:14 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sat Dec 10 23:58:14 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sat Dec 10 23:58:14 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sat Dec 10 23:58:14 2022 kern.info kernel: [  236.365401] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat Dec 10 23:58:14 2022 kern.info kernel: [  236.366552] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Sat Dec 10 23:58:14 2022 kern.info kernel: [  236.367161] 8021q: adding VLAN 0 to HW filter on device eth1
Sat Dec 10 23:58:14 2022 daemon.err insmod: module is already loaded - slhc
Sat Dec 10 23:58:14 2022 daemon.err insmod: module is already loaded - ppp_generic
Sat Dec 10 23:58:14 2022 daemon.err insmod: module is already loaded - pppox
Sat Dec 10 23:58:14 2022 daemon.err insmod: module is already loaded - pppoe
Sat Dec 10 23:58:14 2022 daemon.notice netifd: WAN (7940): uci: Entry not found
Sat Dec 10 23:58:14 2022 daemon.notice netifd: WAN (7940): sh: out of range
Sat Dec 10 23:58:14 2022 daemon.info pppd[7950]: Plugin rp-pppoe.so loaded.
Sat Dec 10 23:58:14 2022 daemon.info pppd[7950]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat Dec 10 23:58:14 2022 daemon.notice pppd[7950]: pppd 2.4.8 started by root, uid 0

1715173329, 2022-12-11:
网卡直通呢？

Karmylr, 2022-12-11:
> 网卡直通呢？

我试试

Karmylr, 2022-12-11:
网卡直通后偶尔会重新拨号。日志：
Sun Dec 11 19:50:12 2022 daemon.notice netifd: Network device 'eth1' link is down
Sun Dec 11 19:50:12 2022 daemon.notice netifd: Interface 'WAN' has link connectivity loss
Sun Dec 11 19:50:12 2022 kern.info kernel: [30438.298129] igb 0000:13:00.0 eth1: igb: eth1 NIC Link is Down
Sun Dec 11 19:50:12 2022 daemon.info pppd[20048]: Terminating on signal 15
Sun Dec 11 19:50:12 2022 daemon.info pppd[20048]: Connect time 30.7 minutes.
Sun Dec 11 19:50:12 2022 daemon.info pppd[20048]: Sent 156083047 bytes, received 2589228012 bytes.
Sun Dec 11 19:50:12 2022 daemon.notice netifd: Network device 'pppoe-WAN' link is down
Sun Dec 11 19:50:13 2022 daemon.err netdata[4109]: Cannot refresh interface pppoe-WAN carrier state by reading '/sys/class/net/pppoe-WAN/carrier'. Stop updating it.
Sun Dec 11 19:50:14 2022 daemon.notice netifd: Network device 'eth1' link is up
Sun Dec 11 19:50:14 2022 daemon.notice netifd: Interface 'WAN' has link connectivity
Sun Dec 11 19:50:14 2022 kern.info kernel: [30439.922384] igb 0000:13:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Sun Dec 11 19:50:15 2022 daemon.notice pppd[20048]: Connection terminated.
Sun Dec 11 19:50:15 2022 daemon.info pppd[20048]: Connect time 30.7 minutes.
Sun Dec 11 19:50:15 2022 daemon.info pppd[20048]: Sent 156083047 bytes, received 2589228012 bytes.
Sun Dec 11 19:50:15 2022 daemon.info pppd[20048]: Sent PADT
Sun Dec 11 19:50:15 2022 daemon.info pppd[20048]: Exit.
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is now down
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Network device 'eth1' link is down
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' has link connectivity loss
Sun Dec 11 19:50:15 2022 kern.info kernel: [30441.406648] 8021q: adding VLAN 0 to HW filter on device eth1
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is now down
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is disabled
Sun Dec 11 19:50:15 2022 daemon.notice netifd: Interface 'WAN' is enabled
Sun Dec 11 19:50:15 2022 kern.info kernel: [30441.470616] 8021q: adding VLAN 0 to HW filter on device eth1
Sun Dec 11 19:50:17 2022 daemon.notice netifd: Network device 'eth1' link is up
Sun Dec 11 19:50:17 2022 daemon.notice netifd: Interface 'WAN' has link connectivity
Sun Dec 11 19:50:17 2022 daemon.notice netifd: Interface 'WAN' is setting up now
Sun Dec 11 19:50:17 2022 kern.info kernel: [30443.146059] igb 0000:13:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Sun Dec 11 19:50:17 2022 daemon.err insmod: module is already loaded - slhc
Sun Dec 11 19:50:17 2022 daemon.err insmod: module is already loaded - ppp_generic
Sun Dec 11 19:50:17 2022 daemon.err insmod: module is already loaded - pppox
Sun Dec 11 19:50:17 2022 daemon.err insmod: module is already loaded - pppoe
Sun Dec 11 19:50:17 2022 daemon.notice netifd: WAN (29160): uci: Entry not found
Sun Dec 11 19:50:17 2022 daemon.notice netifd: WAN (29160): sh: out of range
Sun Dec 11 19:50:17 2022 daemon.info pppd[29170]: Plugin rp-pppoe.so loaded.
Sun Dec 11 19:50:17 2022 daemon.info pppd[29170]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Dec 11 19:50:17 2022 daemon.notice pppd[29170]: pppd 2.4.8 started by root, uid 0
Sun Dec 11 19:50:22 2022 daemon.info pppd[29170]: PPP session is 29483
Sun Dec 11 19:50:22 2022 daemon.warn pppd[29170]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
Sun Dec 11 19:50:22 2022 kern.info kernel: [30448.214274] pppoe-WAN: renamed from ppp0
Sun Dec 11 19:50:22 2022 daemon.info pppd[29170]: Renamed interface ppp0 to pppoe-WAN
Sun Dec 11 19:50:22 2022 daemon.info pppd[29170]: Using interface pppoe-WAN
Sun Dec 11 19:50:22 2022 daemon.notice pppd[29170]: Connect: pppoe-WAN <--> eth1
Sun Dec 11 19:50:23 2022 daemon.err netdata[4109]: Cannot refresh interface pppoe-WAN carrier state by reading '/sys/class/net/pppoe-WAN/carrier'. Stop updating it.
Sun Dec 11 19:50:23 2022 daemon.info netdata[4109]: RRDSET: chart name 'net.pppoe_WAN' on host 'ImmortalWrt' already exists.
Sun Dec 11 19:50:23 2022 daemon.info netdata[4109]: RRDSET: chart name 'net_operstate.pppoe_WAN' on host 'ImmortalWrt' already exists.
Sun Dec 11 19:50:23 2022 daemon.info netdata[4109]: RRDSET: chart name 'net_mtu.pppoe_WAN' on host 'ImmortalWrt' already exists.
Sun Dec 11 19:50:25 2022 daemon.info pppd[29170]: syncppp not active
Sun Dec 11 19:50:25 2022 daemon.info pppd[29170]: Remote message: Authentication success,Welcome!
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: PAP authentication succeeded
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: peer from calling number CC:CC:81:EB:7C:A7 authorized
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: local  LL address fe80::0c27:041b:742c:2ddf
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: remote LL address fe80::cecc:81ff:feeb:7ca7
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: local  IP address 175.42.18.244
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: remote IP address 175.42.16.1
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: primary   DNS address 58.22.96.66
Sun Dec 11 19:50:25 2022 daemon.notice pppd[29170]: secondary DNS address 218.104.128.106
Sun Dec 11 19:50:25 2022 daemon.notice netifd: Network device 'pppoe-WAN' link is up
Sun Dec 11 19:50:25 2022 daemon.notice netifd: Interface 'WAN' is now up

1715173329, 2022-12-11:
不好说，看起来有点像是驱动的问题。

Karmylr, 2022-12-12:
还是会断几秒然后重新拨上号

1715173329, 2022-12-12:
lean 的分支可以正常工作吗？

Karmylr, 2022-12-12:
我试试

Karmylr, 2022-12-12:
lean lede也会断

1715173329, 2022-12-12:
那可能需要考虑下自身因素了 emmm。。。这方面我不是很清楚，推荐去 lean 那边问问

closed 2022-12-12 by Karmylr