immortalwrt/immortalwrt #850: 关于Asus RT-AC88U网络接口识别问题

LtD0gE, 2023-01-01:
immortalwrt对asus RT-AC88U的网络接口识别错误。
AC88U的eth1应为一个BCM53xx交换机芯片，在交换机下分出各个接口。

该机的网络架构应为：交换机芯片的0-3口为LAN1-4，4口为WAN，5口通到RTL8365MB芯片扩展出LAN5-8，eth0、eth2用途不明。而immortalwrt将eth1直接识别为以太网适配器，而不是一个交换机。

根据提取的nvram变量看，原机应该是在eth1下划分两个vlan，其中vlan1负责LAN侧网络，绑定bcm53xx交换机0，1，2，5，7口；vlan2负责WAN侧网络，绑定bcm53xx的4、7口，7口untagged。

希望可以解决一下这个问题，我搞了一晚上vlan没搞通😶

xlighting2017, 2023-01-01:
https://openwrt.org/toh/asus/rt-ac88u

imm这边 现有release 还没有同步到 22.03, 可以试试 master分支(缺web/luci) 或者 openwrt原版的 22.03

LtD0gE, 2023-01-01:
> https://openwrt.org/toh/asus/rt-ac88u
> 
> imm这边 现有release 还没有同步到 22.03, 可以试试 master分支(缺web/luci) 或者 openwrt原版的 22.03

如果在firmware-selector下载是下SNAPSHOT版本吗

1715173329, 2023-01-01:
？现在都是 dsa 了，每个网口都被视为单独的适配器。

1715173329, 2023-01-02:
ref https://www.youtube.com/watch?v=qeuZqRqH-ug

LtD0gE, 2023-01-02:
但我的ac88u的交换机没有识别出DSA

1715173329, 2023-01-02:
这里没有“交换机”。所有的配置在 `网络` -> `接口` -> `设备` 中完成。

1715173329, 2023-01-02:
我没有这个设备，如果有问题，请附图说明。

1715173329, 2023-01-02:
顺便一提，rtl8365mb 的驱动在 kernel 5.16 中才被加入，目前即便是 openwrt 官方主线也没有 backport 这个驱动，extsw 或者说 eth5-8 应该是不可用的。

LtD0gE, 2023-01-02:
[image]

eth1识别成了一个“以太网适配器”，而在原版openwrt和其它教程里这个都是“以太网交换机”。所有识别出的网络接口都列在图上了，lan1-4这些并没有出现。
设备界面配置VLAN只有eth1作为基设备可选，没法绑定到eth1下本应分出的各个端口。

1715173329, 2023-01-02:
贴一下内核日志。

LtD0gE, 2023-01-02:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.225 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r19831-b1833122ea)) #0 SMP Wed Dec 28 06:36:37 2022
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Asus RT-AC88U
[    0.000000] earlycon: ns16550 at MMIO 0x18000300 (options '115200n8')
[    0.000000] printk: bootconsole [ns16550] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Hit pending asynchronous external abort (FSR=0x00001c06) during first unmask, this is most likely caused by a firmware/bootloader bug.
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000]   HighMem zone: 98304 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu s26892 r8192 d22260 u57344
[    0.000000] pcpu-alloc: s26892 r8192 d22260 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130784
[    0.000000] Kernel command line: earlycon
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509704K/524288K available (5833K kernel code, 156K rwdata, 1352K rodata, 1024K init, 292K bss, 14584K reserved, 0K cma-reserved, 393216K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x0a130000 -> 0x3a530000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 16 ways, 256 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x7e530001
[    0.000008] sched_clock: 64 bits at 700MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008107] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xa17102bcf3, max_idle_ns: 440795224838 ns
[    0.019282] Switching to timer-based delay loop, resolution 1ns
[    0.025439] Calibrating delay loop (skipped), value calculated using timer frequency.. 1400.00 BogoMIPS (lpj=7000000)
[    0.036132] pid_max: default: 32768 minimum: 301
[    0.040895] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.048280] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.056618] CPU: Testing write buffer coherency: ok
[    0.061564] CPU0: Spectre v2: using BPIALL workaround
[    0.066937] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.073047] Setting up static identity map for 0x100000 - 0x10003c
[    0.079444] rcu: Hierarchical SRCU implementation.
[    0.084524] smp: Bringing up secondary CPUs ...
[    0.089732] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.089739] CPU1: Spectre v2: using BPIALL workaround
[    0.100647] smp: Brought up 1 node, 2 CPUs
[    0.104779] SMP: Total of 2 processors activated (2800.00 BogoMIPS).
[    0.111168] CPU: WARNING: CPU(s) started in wrong/inconsistent modes (primary CPU mode 0x13)
[    0.119673] CPU: This may indicate a broken bootloader or firmware.
[    0.127996] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.137945] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.144910] pinctrl core: initialized pinctrl subsystem
[    0.150859] NET: Registered protocol family 16
[    0.155862] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.175261] pps_core: LinuxPPS API ver. 1 registered
[    0.180359] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.189608] PTP clock support registered
[    0.198898] clocksource: Switched to clocksource arm_global_timer
[    0.205619] thermal_sys: Registered thermal governor 'step_wise'
[    0.206002] NET: Registered protocol family 2
[    0.216782] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.224984] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.233503] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.241238] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.248364] TCP: Hash tables configured (established 1024 bind 1024)
[... 252 lines cut ...]
[   13.239920] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4366c-pcie for chip BCM43664/4
[   13.248738] brcmfmac 0001:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.asus,rt-ac88u.bin failed with error -2
[   13.263592] brcmfmac 0001:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.asus,rt-ac88u.txt failed with error -2
[... cut ...]
[   19.904002] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   19.910535] br-lan: port 2(wlan0) entered blocking state
[   19.915898] br-lan: port 2(wlan0) entered forwarding state
[   19.921634] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   20.338490] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   20.345006] br-lan: port 1(wlan1) entered blocking state
[   20.350354] br-lan: port 1(wlan1) entered forwarding state
[   28.954704] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x8a, action 0x6
```

LtD0gE, 2023-01-02:
固件来自ImmortalWrt Firmware Selector   没有自定义

xlighting2017, 2023-01-02:
你这是 21.02-SNAPSHOT，   里面 还是 5.4 内核，   官方DSA是 5.10 内核

所以说，
要么 切换到 官方 22.03 (而且还可能有几个口不能用，原因上面说了，缺驱动)， 
要么 上 imm 的 SNAPSHOT（和官方同步，同样也缺驱动）， 而这个没有 Luci，要自己加

总之 这个设备的支持还不成熟， 你要自力更生了

LtD0gE, 2023-01-02:
SNAPSHOT版本在我的机子上跑不起来 :cry: 无限重启，我手头没有tty也抓不到日志

1715173329, 2023-01-02:
看错了，rtl8365mb 在 22.03+ 目前用的是 b53-srab-switch 这个驱动，是可以连通的（但是无法配置）。
不过我弄错了一件事，21.02 还在用 swconfig，也就是传统的交换机配置，这点为之前的言辞道歉。

如果够“幸运”的话，也许能在 `网络` 选项卡下面看到 `交换机` 一栏。
用 swconfig 的话，大概这样配置就好了：
```
--- a/target/linux/bcm53xx/base-files/etc/board.d/02_network
+++ b/target/linux/bcm53xx/base-files/etc/board.d/02_network
@@ -32,7 +32,8 @@ bcm53xx_setup_interfaces()
 			"0:lan" "1:lan" "2:lan" "3:lan" "4:wan" "5t@eth0"
 		;;
 	asus,rt-ac88u)
-		ucidef_set_interfaces_lan_wan "lan1 lan2 lan3 lan4 extsw" "wan"
+		ucidef_add_switch "switch0" \
+			"0:lan:4" "1:lan:3" "2:lan:2" "3:lan:1" "4:wan" "7t@eth1"
 		;;
 	linksys,panamera)
 		ucidef_add_switch "switch1" \
```
不过 21.02 用的 swconfig-b53-srab 似乎并不能让 rtl8365mb 工作，或者需要更多的更改，我没有接触过 bcm53xx。

1715173329, 2023-01-02:
https://github.com/1715173329/imoutowrt/tree/2102-bcm53xx-dsa
如果你可以自己编译的话，试试这个分支。

LtD0gE, 2023-01-03:
该固件未接网线时可看到所有网卡均正常识别，WAN口DHCP及LAN口访问均正常，感谢🙏
这个issue我就先关了，以后再有其它问题的话我再re-open

closed 2023-01-03 by LtD0gE