immortalwrt/immortalwrt #858: 翻译问题 有点玄学

yyhhkya, 2023-01-19:
### Describe the bug

使用github action编译21.02 均没有报错
[https://github.com/yyhhkya/build-immortalwrt](https://github.com/yyhhkya/build-immortalwrt)

之前编译了好几次 这些都全是英文的
[image]

这次新加了上网时间控制 那个统计就变中文了 然后轮到这个新加的变英文了
[image]

### ImmortalWrt version

r19866-5657d4ffb0

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

Newifi-D2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-01-19:
本地缓存问题。

yyhhkya, 2023-01-19:
嗷 抱歉 犯低级错误了~

closed 2023-01-19 by yyhhkya