immortalwrt/immortalwrt #862: Luci错误

King-Of-Knights, 2023-01-23:
### Describe the bug

编译完成后出现以下错误：
```
RPCError
RPC call to luci/getFeatures failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-23.014.25489-a88abc5:11:3)
```

[图片](https://i.328888.xyz/2023/01/23/OT33b.png)

### ImmortalWrt version

Master

### ImmortalWrt target/subtarget

mt7621

### Device

MT1300

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

zxlhhyccc, 2023-01-23:
一样的问题，天灵说make clean

King-Of-Knights, 2023-01-23:
@zxlhhyccc 兄弟好！意思是清除一下就好了吗？

zxlhhyccc, 2023-01-23:
@King-Of-Knights 天灵说make clean，我现在也是在编译，官方更新iwinfo导致的。。。。

King-Of-Knights, 2023-01-23:
@zxlhhyccc 感谢啊，我先试试

zxlhhyccc, 2023-01-23:
@King-Of-Knights 官方也说了make clean，现在官方又更新了。。。
https://github.com/openwrt/openwrt/commit/f3d8de73988a79440d183050dc9711bb29d2a5e1

1715173329, 2023-01-23:
please pull to latest git HEAD (including feeds) and cleanup your buildroot with `make clean` then rebuild.

1715173329, 2023-01-23:
主线默认不带中文，需要自己选择 default-settings-chn。推荐用 openwrt-21.02 分支。

closed 2023-01-23 by 1715173329

zxlhhyccc, 2023-01-23:
> @zxlhhyccc @1715173329 感谢大佬们啊，解决了，顺便问下，为啥目前编译的系统只有英文呢？

你的机器很牛逼啊，这么快就编译好了。。。。

King-Of-Knights, 2023-01-23:
> > @zxlhhyccc @1715173329 感谢大佬们啊，解决了，顺便问下，为啥目前编译的系统只有英文呢？
> 
> 你的机器很牛逼啊，这么快就编译好了。。。。

那倒没有，我是MT1300 小路由器，编译的快
我现在还有个问题，luci-argon-config现在出现了点问题，麻烦大佬帮忙看看：

@zxlhhyccc @1715173329 

```
/usr/lib/lua/luci/ucodebridge.lua:23: ...ib/lua/luci/model/cbi/argon-config/configuration.lua:3: module 'luci.model.ipkg' not found:
no field package.preload['luci.model.ipkg']
no file './luci/model/ipkg.lua'
no file '/usr/share/lua/luci/model/ipkg.lua'
no file '/usr/share/lua/luci/model/ipkg/init.lua'
no file '/usr/lib/lua/luci/model/ipkg.lua'
no file '/usr/lib/lua/luci/model/ipkg/init.lua'
no file './luci/model/ipkg.so'
no file '/usr/lib/lua/luci/model/ipkg.so'
no file '/usr/lib/lua/loadall.so'
no file './luci.so'
no file '/usr/lib/lua/luci.so'
no file '/usr/lib/lua/loadall.so'

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:794:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:795:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:998:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^
```

zxlhhyccc, 2023-01-23:
已经提示了。。。

1715173329, 2023-01-23:
用 21.02 吧，master 不适合日常用途。
我们提供预编译固件和自定义软件包，不需要自行编译。
https://firmware-selector.immortalwrt.org/?version=21.02.3&target=ramips%2Fmt7621&id=glinet_gl-mt1300

King-Of-Knights, 2023-01-23:
@1715173329 @zxlhhyccc 感谢这么及时的回答，好像是ipkg没装，我再抽时间看看👀