immortalwrt/immortalwrt #864: Can not install luci

triatomic, 2023-01-27:
### Describe the bug

```
root@ImmortalWrt:~# opkg install luci
Installing luci (git-22.313.56929-b20fef3) to root...
Downloading https://downloads.immortalwrt.org/snapshots/packages/aarch64_cortex-a53/luci/luci_git-22.313.56929-b20fef3_all.ipk
Installing luci-proto-ipv6 (git-21.148.48881-79947af) to root...
Downloading https://downloads.immortalwrt.org/snapshots/packages/aarch64_cortex-a53/luci/luci-proto-ipv6_git-21.148.48881-79947af_all.ipk
Collected errors:
 * opkg_download: Failed to download https://downloads.immortalwrt.org/snapshots/packages/aarch64_cortex-a53/luci/luci-proto-ipv6_git-21.148.48881-79947af_all.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download luci-proto-ipv6. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci.
```

### ImmortalWrt version

r25105-6b1e639bfc

### ImmortalWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-01-27:
feeds are broken due to network issue on our buuldbot.
It should be fixed in next few hours.

1715173329, 2023-01-27:
fixed now.

closed 2023-01-27 by 1715173329