immortalwrt/immortalwrt #866: 红米ax6s刷入 SNAPSHOT Sysupgrade后黄灯闪烁无法启动

1368129224, 2023-01-29:
### Describe the bug

1. 刷入官方内测固件1.2.7开启ssh
2. ssh刷入Factory 映像 `mtd -r write /tmp/factory.bin firmware`
3. 自动重启后进入web页面，通过luci选择Sysupgrade 映像升级
4. 自动重启后led灯显示黄色，设备无法启动

### ImmortalWrt version

SNAPSHOT (r25105-6b1e639bfc)

### ImmortalWrt target/subtarget

 mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S / Xiaomi Router AX3200

### Image kind

Official downloaded image

### Steps to reproduce

1. 刷入官方内测固件1.2.7开启ssh
2. ssh刷入Factory 映像 `mtd -r write /tmp/factory.bin firmware`
3. 自动重启后进入web页面，通过luci选择Sysupgrade 映像升级
4. 自动重启后led灯显示黄色，设备无法启动

### Actual behaviour

设备无法启动

### Expected behaviour

设备正常启动并正常使用

### Additional info

镜像从官方网站下载，设备购于22年9月底

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-01-29:
需要接 TTL 捕捉日志，无法算命。

1368129224, 2023-01-29:
后续提供日志，先关闭issue

closed 2023-01-29 by 1368129224

zxlhhyccc, 2023-01-29:
> 后续提供日志，先关闭issue

砖了，50收砖。。。。

1715173329, 2023-01-29:
？

zxlhhyccc, 2023-01-29:
> ？

天灵，红米ax6s性价比咋样？

1715173329, 2023-01-29:
这里不是聊天的地方，不要发送不相关且无意义的回复。

tangfenga, 2023-01-29:
@zxlhhyccc 我也遇到过相同的问题，我已经解决了，请稍等我找一下当时所做的更改😉

1368129224, 2023-01-30:
> @zxlhhyccc 我也遇到过相同的问题，我已经解决了，请稍等我找一下当时所做的更改😉

Hi 请问你有遇到过刷入后黄灯的问题吗？请问怎么解决，谢谢

tangfenga, 2023-01-30:
@1368129224 这个bug在openwrt官方仓库22.03已经修复但是我忘记当时openwrt论坛中修复这个问题的答案了

tangfenga, 2023-01-30:
@tangfenga 我目前factory.bin以及sysgrade都用的openwrt官方主线版本，没有出现问题，可以先替代一下，等我找到当时diff在哪之后再发到issue里😘