immortalwrt/immortalwrt #873: 希望能添加2个内核模块的编译

qiang-yu, 2023-02-16:
### Describe the bug

kmod-nf-tproxy  
kmod-nft-tproxy 

这两个内核模块没有编译 

目前 OpenWrt 透明代理 tcp/udp 的方式

TCP : Redirect  或者  TProxy (kmod-nf-tproxy  kmod-nft-tproxy)
UDP : TProxy (kmod-ipt-tproxy)

现在的编译中，kmod-ipt-tproxy 是有的，所以  UDP 可以走 TProxy 透明代理
但是缺少 kmod-nf-tproxy  kmod-nft-tproxy 两个模块， TCP 没法走 TProxy 

另外，ipv6 的透明代理也必须要这两个模块

希望编译能加上这两个内核模块

谢谢

### ImmortalWrt version

21.02.4

### ImmortalWrt target/subtarget

All

### Device

All

### Image kind

Self-built image

### Steps to reproduce

opkg update
opkg list | grep tproxy

可以看到没有这两个模块可用，应该是没有编译这两个模块

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-02-16:
明确一点，kmod-nft-tproxy 是 nftables 的 tproxy 实现。
21.02 使用 iptables，kmod-ipt-tproxy 是对应的 tproxy 实现。
这与 TCP、UDP 或 IPv6 没有任何关系。

如果你的环境中 TPROXY 出现问题，请检查你自己的防火墙规则。

qiang-yu, 2023-02-16:
好的，那我检查一下，谢谢

closed 2023-02-16 by qiang-yu