immortalwrt/immortalwrt #878: MT7620 snapshot，似乎遇到USB相关的东西，负载一上去就会重启

machine-doll, 2023-02-18:
### Describe the bug

设备是newifi mini，代码是snaoshot，最近这个月测试过好几次了，snapsho和21.02.3都试过  
试过只编译进去kmod-usb-net-rdnis，但是接手机的USB网络共享，最多过几分钟就会重启
试过编译进去usb存储方面的驱动，发现如果不挂载文件系统还好，但是只要把文件系统的驱动编译进去，并且挂载了，最多只有刚挂载的那段时间是正常的，以后只要一挂载就会重启  
一开始怀疑过供电不够，怀疑过硬件损坏，但是发现如果改用padavan是没问题的，不过padavan实在是太老了  
另外不知道是不是我这边才会有的问题，一些使用SPI NOR的MT7621机器，比如Mi4A千兆版，开机过程中就会重启，不过这个问题原版也有，应该是上游不知道什么时候引入的问题

### ImmortalWrt version

r25302-e88be07d0d

### ImmortalWrt target/subtarget

ramips

### Device

lenovo y1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-02-18:
TTL 日志。

machine-doll, 2023-08-09:
> TTL 日志。
好久没用那个机器，上个月翻出来编译了一下，发现好了
SPI NOR的问题暂时没有再遇到，手里SPI NOR的7621机器就只有e8820v2了

closed 2023-08-09 by machine-doll