immortalwrt/immortalwrt #879: AC2100最新版固件似乎无法启动

U2FsdGVkX1, 2023-02-19:
### Describe the bug

目前21.02.3是正常的，21.02.4版本直接无法启动，全程灯没有闪烁，也没有分配到IP，应该是没有进入系统
暂时没有TTL线拿日志，有设备的可以帮忙测试补充一下？

### ImmortalWrt version

21.02.4

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Redmi Router AC2100

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

f0re1gnKey, 2023-02-22:
K2P在21.02.4也无法启动，具体表现为系统启动正常，但网口全部不通，刷会21.02.3则一切正常

1715173329, 2023-02-23:
fixed by c3b16df57fb9a63e315b971317ad864e75dc82ff

closed 2023-02-23 by 1715173329

sqnight-mm6, 2023-02-23:
VR500在21.02.4也无法启动，刷回21.02.3正常，也是MT7621

U2FsdGVkX1, 2023-02-23:
> fixed by [c3b16df](https://github.com/immortalwrt/immortalwrt/commit/c3b16df57fb9a63e315b971317ad864e75dc82ff)

ImmortalWrt官网的ramips stable镜像可以重新编译一下吗

1715173329, 2023-02-23:
21.02-SNAPSHOT 的包会在三个小时后更新，固定 release 需要等 21.02.5 释出。