immortalwrt/immortalwrt #886: docker拉取镜像失败

Leslie-Luo, 2023-02-24:
### Describe the bug

自己编译的固件内置了docker，使用docker拉取镜像时报错

```
root@ImmortalWrt:~# docker pull jellyfin/jellyfin
Using default tag: latest
latest: Pulling from jellyfin/jellyfin
8740c948ffd4: Downloading 
e318c113e9bf: Download complete 
28f984869f48: Download complete 
9f168e85cffa: Download complete 
latest: Pulling from jellyfin/jellyfin
8740c948ffd4: Pull complete 
e318c113e9bf: Extracting [==================================================>]  280.1MB/280.1MB
28f984869f48: Download complete 
9f168e85cffa: Download complete 
failed to register layer: operation not supported
```
[image]

docker版本信息
```
root@ImmortalWrt:~# docker version
Client:
 Version:           20.10.22
 API version:       1.41
 Go version:        go1.18.10
 Git commit:        3a2c30b
 Built:             Fri Feb 24 10:01:48 2023
 OS/Arch:           linux/amd64
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.22
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.18.10
  Git commit:       42c8b31
  Built:            Thu Feb 23 03:55:55 2023
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.6.13
  GitCommit:        
 runc:
  Version:          1.1.4
  GitCommit:        
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

### ImmortalWrt version

r19903-b4f56cda2b

### ImmortalWrt target/subtarget

x86/64

### Device

CncTion J4125-4L-I225

### Image kind

Self-built image

### Steps to reproduce

安装我自编译的镜像：
https://github.com/Leslie-Luo/openwrt-firmware/releases/download/2023.02.24_19-46/immortalwrt-x86-64-generic-squashfs-combined.img.gz
在终端执行
`docker pull jellyfin/jellyfin`

### Actual behaviour

报错无法拉取镜像

### Expected behaviour

成功拉取镜像

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DOCKER_CGROUP_OPTIONS=y
CONFIG_DOCKER_NET_MACVLAN=y
CONFIG_DOCKER_STO_EXT4=y
# CONFIG_GRUB_IMAGES is not set
CONFIG_GRUB_TIMEOUT="0"
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_NET_PRIO=y
CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y
CONFIG_KERNEL_EXT4_FS_SECURITY=y
CONFIG_KERNEL_FS_POSIX_ACL=y
CONFIG_KERNEL_NET_CLS_CGROUP=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_PACKAGE_adbyby=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgroupfs-mount=y
CONFIG_PACKAGE_containerd=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-services=y
CONFIG_PACKAGE_docker=y
CONFIG_PACKAGE_dockerd=y
CONFIG_PACKAGE_filebrowser=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_kmod-br-netfilter=y
CONFIG_PACKAGE_kmod-crypto-acompress=y
CONFIG_PACKAGE_kmod-dax=y
CONFIG_PACKAGE_kmod-dm=y
CONFIG_PACKAGE_kmod-dummy=y
CONFIG_PACKAGE_kmod-fs-btrfs=y
CONFIG_PACKAGE_kmod-ikconfig=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
CONFIG_PACKAGE_kmod-lib-lzo=y
CONFIG_PACKAGE_kmod-lib-raid6=y
CONFIG_PACKAGE_kmod-lib-xor=y
CONFIG_PACKAGE_kmod-lib-zlib-deflate=y
CONFIG_PACKAGE_kmod-lib-zlib-inflate=y
CONFIG_PACKAGE_kmod-lib-zstd=y
CONFIG_PACKAGE_kmod-nf-ipvs=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-veth=y
CONFIG_PACKAGE_libattr=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcap-bin=y
CONFIG_PACKAGE_libcap-bin-capsh-shell="/bin/sh"
CONFIG_PACKAGE_libcap-ng=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libdb47=y
CONFIG_PACKAGE_libdevmapper=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libmount=y
CONFIG_PACKAGE_libnetwork=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libruby=y
CONFIG_PACKAGE_libseccomp=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_libyaml=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-argon-config=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-docker=y
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-smartdns=y
CONFIG_PACKAGE_luci-app-socat=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-uhttpd=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-i18n-adbyby-plus-zh-cn=y
CONFIG_PACKAGE_luci-i18n-argon-config-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ddns-zh-cn=y
CONFIG_PACKAGE_luci-i18n-diskman-zh-cn=y
CONFIG_PACKAGE_luci-i18n-docker-zh-cn=y
CONFIG_PACKAGE_luci-i18n-dockerman-zh-cn=y
CONFIG_PACKAGE_luci-i18n-netdata-zh-cn=y
CONFIG_PACKAGE_luci-i18n-smartdns-zh-cn=y
CONFIG_PACKAGE_luci-i18n-socat-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-uhttpd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-upnp-zh-cn=y
CONFIG_PACKAGE_luci-lib-docker=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_miniupnpd=y
CONFIG_PACKAGE_mount-utils=y
CONFIG_PACKAGE_netdata=y
CONFIG_PACKAGE_parted=y
CONFIG_PACKAGE_ruby=y
CONFIG_PACKAGE_ruby-bigdecimal=y
CONFIG_PACKAGE_ruby-date=y
CONFIG_PACKAGE_ruby-dbm=y
CONFIG_PACKAGE_ruby-digest=y
CONFIG_PACKAGE_ruby-enc=y
CONFIG_PACKAGE_ruby-forwardable=y
CONFIG_PACKAGE_ruby-pstore=y
CONFIG_PACKAGE_ruby-psych=y
CONFIG_PACKAGE_ruby-stringio=y
CONFIG_PACKAGE_ruby-strscan=y
CONFIG_PACKAGE_ruby-yaml=y
CONFIG_PACKAGE_runc=y
CONFIG_PACKAGE_smartdns=y
CONFIG_PACKAGE_smartmontools=y
CONFIG_PACKAGE_socat=y
CONFIG_PACKAGE_tini=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_uclibcxx=y
CONFIG_PACKAGE_unzip=y
CONFIG_PARTED_READLINE=y
CONFIG_TARGET_KERNEL_PARTSIZE=512
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
# CONFIG_TARGET_ROOTFS_TARGZ is not set
# CONFIG_PACKAGE_kmod-inet-diag is not set
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

sbwml, 2023-02-27:
docker 它不应该工作在 overlayfs 分区，docker 本身就会创建 overlayfs ，如果用套中套方式它肯定是会出问题的，内核它也不允许在那个文件系统上这样干。

大佬应该新建一个分区并格式化成ext4、xfs或btrfs 这些常见文件系统并设置为docker 工作目录。如果条件不允许大佬这样做，大佬还能直接使用 ext4 文件系统的固件，都能解决大佬上面遇到的小问题。

Leslie-Luo, 2023-02-27:
> docker 它不应该工作在 overlayfs 分区，docker 本身就会创建 overlayfs ，如果用套中套方式它肯定是会出问题的，内核它也不允许在那个文件系统上这样干。
> 
> 大佬应该新建一个分区并格式化成ext4、xfs或btrfs 这些常见文件系统并设置为docker 工作目录。如果条件不允许大佬这样做，大佬还能直接使用 ext4 文件系统的固件，都能解决大佬上面遇到的小问题。

请问大佬在编译时如何为docker设置分区呢？

sbwml, 2023-02-27:
我不知道，但是大佬肯定是可以在现有的硬盘上用空闲位置新建一个分区给docker 用。

closed 2023-03-04 by 1715173329

reply2future, 2024-02-28:
> > docker 它不应该工作在 overlayfs 分区，docker 本身就会创建 overlayfs ，如果用套中套方式它肯定是会出问题的，内核它也不允许在那个文件系统上这样干。
> > 大佬应该新建一个分区并格式化成ext4、xfs或btrfs 这些常见文件系统并设置为docker 工作目录。如果条件不允许大佬这样做，大佬还能直接使用 ext4 文件系统的固件，都能解决大佬上面遇到的小问题。
> 
> 请问大佬在编译时如何为docker设置分区呢？

Maybe you could follow [these steps](https://github.com/mwarning/docker-openwrt-build-env?tab=readme-ov-file#usage-macosx)