immortalwrt/immortalwrt #889: K2P(mt7621) 2.4G wifi 无法开启256QAM(vht)

f0re1gnKey, 2023-02-26:
### Describe the bug

具体设置为 uci set wireless.radio0.vendor_vht='1'
无线无法启动

开启后系统日志报错
daemon.err hostapd: Configured VHT capability [VHT_CAP_SUPP_CHAN_WIDTH_MASK] exceeds max value supported by the driver (2 > 0)
daemon.err hostapd: Interface initialization failed
内核日志无报错

如果不开启的话，winfi显示是802.11n第四代，且速率显示600m（不正常）
其他正常的固件可以开启，并显示802.11ac第五代，且速率显示300m（正常）

系统固件使用的是21.02-snapshot，因为21.02.4在mt7621上无法成功启动，这个原因具体看
https://github.com/immortalwrt/immortalwrt/issues/879

### ImmortalWrt version

r19903-b4f56cda2b

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

Phicomm K2P

### Image kind

Official downloaded image

### Steps to reproduce

无线设置->编辑->上方高级设置->启用 256-QAM

### Actual behaviour

无法启动。如关闭，则winfi显示802.11n的2.4G频段

### Expected behaviour

成功启用，无线正常，winfi显示802.11ac的2.4G频段

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-02-26:
```patch
diff --git a/package/kernel/mt76/patches/010-mt76-allow-VHT-rate-on-2.4GHz.patch b/package/kernel/mt76/patches/010-mt76-allow-VHT-rate-on-2.4GHz.patch
index 2661848828..23c4d4685c 100644
--- a/package/kernel/mt76/patches/010-mt76-allow-VHT-rate-on-2.4GHz.patch
+++ b/package/kernel/mt76/patches/010-mt76-allow-VHT-rate-on-2.4GHz.patch
@@ -55,3 +55,14 @@ Signed-off-by: DENG Qingfang <dqfext@gmail.com>
  		if (ret)
  			return ret;
  	}
+--- a/mt7615/init.c
++++ b/mt7615/init.c
+@@ -409,6 +409,8 @@ mt7615_init_wiphy(struct ieee80211_hw *h
+ 		hw->max_tx_fragments = MT_HW_TXP_MAX_BUF_NUM;
+
+ 	phy->mt76->sband_2g.sband.ht_cap.cap |= IEEE80211_HT_CAP_LDPC_CODING;
++	phy->mt76->sband_2g.sband.vht_cap.cap |=
++			IEEE80211_VHT_CAP_MAX_A_MPDU_LENGTH_EXPONENT_MASK;
+ 	phy->mt76->sband_5g.sband.ht_cap.cap |= IEEE80211_HT_CAP_LDPC_CODING;
+ 	phy->mt76->sband_5g.sband.vht_cap.cap |=
+ 			IEEE80211_VHT_CAP_MAX_A_MPDU_LENGTH_EXPONENT_MASK;
```

apply this patch and check if it works

f0re1gnKey, 2023-02-26:
感谢您的答复
patch的时候似乎出了点状况，为了测试暂且先手动贴了上去

/package/kernel/mt76/patches/010-mt76-allow-VHT-rate-on-2.4GHz.patch
`From 62468f0b5a72e506915cf50176b701626767e353 Mon Sep 17 00:00:00 2001
From: DENG Qingfang <dengqf6@mail2.sysu.edu.cn>
Date: Mon, 23 Nov 2020 10:46:37 +0800
Subject: [PATCH] mt76: allow VHT rate on 2.4GHz

Allow chips that support 11ac to use 256QAM on 2.4GHz

---
 mac80211.c | 10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

--- a/mac80211.c
+++ b/mac80211.c
@@ -281,7 +281,7 @@ static void mt76_init_stream_cap(struct
 void mt76_set_stream_caps(struct mt76_phy *phy, bool vht)
 {
 	if (phy->cap.has_2ghz)
-		mt76_init_stream_cap(phy, &phy->sband_2g.sband, false);
+		mt76_init_stream_cap(phy, &phy->sband_2g.sband, vht);
 	if (phy->cap.has_5ghz)
 		mt76_init_stream_cap(phy, &phy->sband_5g.sband, vht);
 	if (phy->cap.has_6ghz)
@@ -348,13 +348,13 @@ mt76_init_sband(struct mt76_phy *phy, st
 
 static int
 mt76_init_sband_2g(struct mt76_phy *phy, struct ieee80211_rate *rates,
-		   int n_rates)
+		   int n_rates, bool vht)
 {
 	phy->hw->wiphy->bands[NL80211_BAND_2GHZ] = &phy->sband_2g.sband;
 
 	return mt76_init_sband(phy, &phy->sband_2g, mt76_channels_2ghz,
 			       ARRAY_SIZE(mt76_channels_2ghz), rates,
-			       n_rates, true, false);
+			       n_rates, true, vht);
 }
 
 static int
@@ -506,7 +506,7 @@ int mt76_register_phy(struct mt76_phy *p
 		return ret;
 
 	if (phy->cap.has_2ghz) {
-		ret = mt76_init_sband_2g(phy, rates, n_rates);
+		ret = mt76_init_sband_2g(phy, rates, n_rates, vht);
 		if (ret)
 			return ret;
 	}
@@ -644,7 +644,7 @@ int mt76_register_device(struct mt76_dev
 		return ret;
 
 	if (phy->cap.has_2ghz) {
-		ret = mt76_init_sband_2g(phy, rates, n_rates);
+		ret = mt76_init_sband_2g(phy, rates, n_rates, vht);
 		if (ret)
 			return ret;
 	}

--- a/mt7615/init.c
+++ b/mt7615/init.c
@@ -409,6 +409,8 @@ mt7615_init_wiphy(struct ieee80211_hw *h
 		hw->max_tx_fragments = MT_HW_TXP_MAX_BUF_NUM;

 	phy->mt76->sband_2g.sband.ht_cap.cap |= IEEE80211_HT_CAP_LDPC_CODING;
+	phy->mt76->sband_2g.sband.vht_cap.cap |=
+			IEEE80211_VHT_CAP_MAX_A_MPDU_LENGTH_EXPONENT_MASK;
 	phy->mt76->sband_5g.sband.ht_cap.cap |= IEEE80211_HT_CAP_LDPC_CODING;
 	phy->mt76->sband_5g.sband.vht_cap.cap |=
 			IEEE80211_VHT_CAP_MAX_A_MPDU_LENGTH_EXPONENT_MASK;`

git pull了最新源码，问题依然存在

1715173329, 2023-03-13:
这个是 mt76 驱动本身不支持，上游对此也不感兴趣。
其他“正常的固件”能用，是因为它们使用的是 mtk 原厂闭源（泄露）无线驱动。

我不认为折腾 2.4G 有什么实际意义，或者您也可以直接用能用的固件。

closed 2023-03-13 by 1715173329