immortalwrt/immortalwrt #891: 成品稳定固件ImmortalWrt 21.02.4 无法启动passwall

baodp, 2023-03-01:
### Describe the bug

成品稳定固件ImmortalWrt 21.02.4 无法启动passwall。
显示需要安装
[image]
但是我系统查了有这个包了。
请问怎么解决还是不支持？？

主要我前几天被踢出TG群了只能这里问问。

### ImmortalWrt version

21.02.4 

### ImmortalWrt target/subtarget

21.02.4 

### Device

X86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-03-01:
passwall 的依赖写得有问题，需要自行安装。

closed 2023-03-01 by 1715173329

baodp, 2023-03-01:
[image]
需要哪些依赖呢？我菜。。

baodp, 2023-03-01:
SSRP+也是运行不了。
openclash  确可以运行。

hehe9011, 2023-03-02:
SSRP+我的也不行

nantayo, 2023-03-12:
我查看库里自带的luci-app-passwall的[makefile](https://github.com/immortalwrt/luci/blob/master/applications/luci-app-passwall/Makefile#L64)文件，发现缺失了关于iptables和nftables的判断，你可以跟[我维护的](https://github.com/nantayo/passwall/blob/master/luci-app-passwall/Makefile#L63-L86)或者[xiaorouji的原版](https://github.com/xiaorouji/openwrt-passwall/blob/luci/luci-app-passwall/Makefile#L66-L89)对比一下，很明显能看出来缺了一段，这导致很多相关依赖是没有安装上的。
***
解决办法要么你手动把缺失的那部分依赖安装上，要么用：
rm -rf ./feeds/luci/applications/luci-app-passwall
rm -rf ./package/feeds/luci/luci-app-passwall
这两句删去原本自带的passwall，然后svn co我的或者xiaorouji的passwall，编译出来就是正常的了。
***
顺便一提，一般情况下我建议你用xiaorouji的，人家是原版，我的这份fork是为了解决v2ray更新之后删去alterid导致旧机场没法用的问题。如果你的机场或者节点没用VMESS，或者已经更新了的话还是用原版吧。

nantayo, 2023-03-12:
> > SSRP+也是运行不了。 openclash 确可以运行。
> 
> 编译的时候需要选中ipset

是库里自带passwall的makefile有问题，跟原版比缺了一段。

ksong008, 2023-03-12:
@nantayo 21.02，master分清楚了？码字太累

nantayo, 2023-03-12:
> @nantayo 21.02，master分清楚了？码字太累

看走眼了，不过master和21.02分支改makefile是同样的改法啊，都是直接把iptables的依赖写在一开始的depends里。至于删去了几个依赖，就不清楚是不是21.02没有或者用不上了，但确实是少了几个

ksong008, 2023-03-12:
master 默认fw4，不需要iptables