immortalwrt/immortalwrt #892: mlx4网卡本地禁用再启用后无法连接

killsky, 2023-03-04:
### Describe the bug

无论master还是退回21.02，已选择mii、mlx4、5的驱动，加载后将网卡加入桥接。
本地pc只有重启后可以正常连接，只要禁用再启用，本地连接就只有发送没有接收。
并且openwrt网口信息显示该端口已断开，只有重启过后才可以使用。
硬件：r86s 
cx342万兆网卡

### ImmortalWrt version

r25429-db5e542439

### ImmortalWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware20,1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

killsky, 2023-03-05:
补充了一些日志和截图，望哪位大佬解答一下。换了R86S路由就这个问题最头疼，实在是不想放弃immortal的易用性和先进性。

1715173329, 2023-03-17:
你可以试下最新的 21.02-snapshot，或者编译 master 的时候自己打个补丁
https://transfer.sh/ku7HQu/100-Revert-device-add-support-for-configuring-device-link-spe.patch
放到 package/network/config/netifd/patches 下面

我们的配置都是跟随 OpenWrt 上游走的，如果不行的话就需要问上游了。

killsky, 2023-03-17:
感谢，一直在使用最新的master编译，马上加补丁测试。感谢回复！

killsky, 2023-03-18:
试过补丁和21.02，都没有问题了。感谢！

closed 2023-03-18 by killsky