immortalwrt/immortalwrt #893: 默认编译的x86_64，web管理界面打不开。

baodp, 2023-03-04:
### Describe the bug

默认编译的x86_64，就添加了homeproxy。网卡是i350双口。之前下载你们编译好的版本没有问题。
web管理界面打不开，shh登录正常，正常。网口都试过换着插了，也不行。

PS:  你们放出来的固件也是正常使用的。大雕的固件是正常的。

编译命令的方式是使用大雕源那种命令进行的。只是改套用你们的源地址和CD目录。

### ImmortalWrt version

r25436-5da18713f3

### ImmortalWrt target/subtarget

x86/64

### Device

INTEL HM570

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

jh0072110, 2023-03-04:
master分支luci不是默认选项，编译的时候需要选中

baodp, 2023-03-04:
> master分支luci不是默认选项，编译的时候需要选中

谢谢。

closed 2023-03-04 by baodp