immortalwrt/immortalwrt #900: mtl7621 r6900v2 无法启动.

lhuanyun, 2023-03-07:
### Describe the bug

**下载地址**：https://firmware-selector.immortalwrt.org/?version=21.02.4&target=ramips%2Fmt7621&id=netgear_r6900-v2

```
型号
NETGEAR R6900 v2
平台
ramips/mt7621
版本
SNAPSHOT (r25429-db5e542439)
日期
2023-03-03 04:35:28
```

```
型号
NETGEAR R6900 v2
平台
ramips/mt7621
版本
21.02.4 (r19892-a6dce2ceb7)
日期
2023-02-14 09:20:45
```

这两个版本刷入无法启动，测试openwrt官版正常使用。
请老大检查，谢谢。

### ImmortalWrt version

21.02.4

### ImmortalWrt target/subtarget

21.02.4

### Device

r6900v2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

closed 2023-03-07 by 1715173329

lhuanyun, 2023-03-07:
```
型号
NETGEAR R6900 v2
平台
ramips/mt7621
版本
SNAPSHOT (r25429-db5e542439)
日期
2023-03-03 04:35:28
```

snapshot 版本也无法正常启动

1715173329, 2023-03-07:
snapshot 没 luci 就等于无法启动了？

lhuanyun, 2023-03-07:
> 

谢谢老大的回复，我的r6900v2 没有分配到ip，应该是没有启动成功吧。
请问目前r6900v2 ，可以使用的版本是多少呢。感谢辛苦的工作

1715173329, 2023-03-07:
推荐官方。

lhuanyun, 2023-03-07:
可是很喜欢[immortalwrt](https://github.com/immortalwrt/immortalwrt)这个项目啊，它更适合国人使用

1715173329, 2023-03-07:
那你就给 TTL 日志。没有 TTL 某宝 ch340g 三块钱包邮到家。

lhuanyun, 2023-03-07:
> 那你就给 TTL 日志。没有 TTL 某宝 ch340g 三块钱包邮到家。

有ch340g ，没有t10螺丝刀，没办法打开r6900天灵盖....

1715173329, 2023-03-07:
没日志那只能去街上请个算命先生给你算算为什么不能启动。

lhuanyun, 2023-03-07:
> 没日志那只能去街上请个算命先生给你算算为什么不能启动。

lol 谢谢大佬咯，有时间我再看看咧