immortalwrt/immortalwrt #905: passwall  安装后启动不成功，试过取消勾选然后勾选进程守护的方法，一样不行，系统是新装的

zzduci, 2023-03-15:
### Describe the bug

passwall  安装后启动不成功，试过取消勾选然后勾选进程守护的方法，一样不行，系统是新装的  
[image]
[image]
环境为 X86 PVE下虚拟机安装

### ImmortalWrt version

r19892-a6dce2ceb7

### ImmortalWrt target/subtarget

r19892-a6dce2ceb7

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

qm importdisk 命令导入immortalwrt-21.02.4-x86-64-generic-squashfs-combined.img 这个镜像，直接启动成功后，设置好网络后，直接安装passwall,   网络环境openwrt准备做旁路由 ，安装成功passwall后启动不成功，显示TCP未运行，UDP未运行

### Actual behaviour

qm importdisk 命令导入immortalwrt-21.02.4-x86-64-generic-squashfs-combined.img 这个镜像，直接启动成功后，设置好网络后，直接安装passwall,   网络环境openwrt准备做旁路由 ，安装成功passwall后启动不成功，显示TCP未运行，UDP未运行

### Expected behaviour

qm importdisk 命令导入immortalwrt-21.02.4-x86-64-generic-squashfs-combined.img 这个镜像，直接启动成功后，设置好网络后，直接安装passwall,   网络环境openwrt准备做旁路由 ，安装成功passwall后启动不成功，显示TCP未运行，UDP未运行

### Additional info

qm importdisk 命令导入immortalwrt-21.02.4-x86-64-generic-squashfs-combined.img 这个镜像，直接启动成功后，设置好网络后，直接安装passwall,   网络环境openwrt准备做旁路由 ，安装成功passwall后启动不成功，显示TCP未运行，UDP未运行

### Diffconfig

```text
无
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

zzduci, 2023-03-15:
https://immortalwrt.kyarucloud.moe/releases/21.02.0/targets/x86/64/immortalwrt-21.02.0-x86-64-generic-squashfs-combined.img.gz  用的官方镜像

1715173329, 2023-03-15:
重启即可，重载 ucitrack。

1715173329, 2023-03-15:
顺便一提，passwall 并没有按标准写入透明代理相关的依赖，你可能需要自己安装。

closed 2023-03-15 by 1715173329

zzduci, 2023-03-15:
> 重载 ucitrack

怎么重载，是删了重装吗？

1715173329, 2023-03-15:
重启

zzduci, 2023-03-15:
> ucitrack

不行还是一样的，能试的方法都试过了，才到这里来反馈的，我也试过   “执行重置”，再次安装passwall,还是同样的问题

1715173329, 2023-03-15:
那就是你的依赖问题咯，你自己看日志啊

zzduci, 2023-03-15:
[image]
什么f都没有看安装，就是初始状态
 ImmortalWrt 21.02.4, r19892-a6dce2ceb7
 -----------------------------------------------------
root@ImmortalWrt:~# opkg install luci-app-passwall
Installing luci-app-passwall (4.59) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/luci/luci-app-passwall_4.59_all.ipk
Installing coreutils (8.32-6) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/coreutils_8.32-6_x86_64.ipk
Installing coreutils-base64 (8.32-6) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/coreutils-base64_8.32-6_x86_64.ipk
Installing coreutils-nohup (8.32-6) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/coreutils-nohup_8.32-6_x86_64.ipk
Installing libnghttp2-14 (1.43.0-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/libnghttp2-14_1.43.0-1_x86_64.ipk
Installing libcurl4 (7.83.1-18.1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/libcurl4_7.83.1-18.1_x86_64.ipk
Installing curl (7.83.1-18.1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/curl_7.83.1-18.1_x86_64.ipk
Installing dns2socks (2.1-2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/dns2socks_2.1-2_x86_64.ipk
Installing dns2tcp (1.1.0-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/dns2tcp_1.1.0-1_x86_64.ipk
Installing libelf1 (0.180-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/libelf1_0.180-1_x86_64.ipk
Installing libbpf0 (5.10.10-2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/libbpf0_5.10.10-2_x86_64.ipk
Installing ip-full (5.11.0-3) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/ip-full_5.11.0-3_x86_64.ipk
Installing libuci-lua (2021-04-14-4b3db117-5) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/libuci-lua_2021-04-14-4b3db117-5_x86_64.ipk
Installing microsocks (1.0.3-3) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/microsocks_1.0.3-3_x86_64.ipk
Installing resolveip (2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/resolveip_2_x86_64.ipk
Installing tcping (0.3-2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/tcping_0.3-2_x86_64.ipk
Installing unzip (6.0-8) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/unzip_6.0-8_x86_64.ipk
Installing libltdl7 (2.4.6-2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/libltdl7_2.4.6-2_x86_64.ipk
Installing liblua5.3-5.3 (5.3.5-4) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/liblua5.3-5.3_5.3.5-4_x86_64.ipk
Installing libreadline8 (8.1-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/libreadline8_8.1-1_x86_64.ipk
Installing libatomic1 (8.4.0-3) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/targets/x86/64/packages/libatomic1_8.4.0-3_x86_64.ipk
Installing haproxy (2.2.24-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/haproxy_2.2.24-1_x86_64.ipk
Installing libev (4.31-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/libev_4.31-1_x86_64.ipk
Installing libmbedtls12 (2.16.12-21) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/base/libmbedtls12_2.16.12-21_x86_64.ipk
Installing libsodium (1.0.18-2022-02-27-2e9b4c47-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/libsodium_1.0.18-2022-02-27-2e9b4c47-1_x86_64.ipk
Installing shadowsocks-libev-config (3.3.5-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/shadowsocks-libev-config_3.3.5-1_x86_64.ipk
Installing shadowsocks-libev-ss-local (3.3.5-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/shadowsocks-libev-ss-local_3.3.5-1_x86_64.ipk
Installing shadowsocks-libev-ss-redir (3.3.5-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/shadowsocks-libev-ss-redir_3.3.5-1_x86_64.ipk
Installing libcares (1.18.1-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/libcares_1.18.1-1_x86_64.ipk
Installing shadowsocks-libev-ss-server (3.3.5-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/shadowsocks-libev-ss-server_3.3.5-1_x86_64.ipk
Installing libudns (0.4-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/libudns_0.4-1_x86_64.ipk
Installing shadowsocksr-libev-ssr-local (2.5.6-9) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/21.02.4/packages/x86_64/packages/shadowsocksr-libev-ssr-local_2.5.6-9_x86_64.ipk
[... cut ...]
Configuring unzip.
Configuring libnghttp2-14.
Configuring libcurl4.
Configuring curl.
Configuring microsocks.
Configuring shadowsocks-libev-ss-local.
Configuring simple-obfs-client.
Configuring libudns.
Configuring shadowsocksr-libev-ssr-local.
Configuring libcares.
Configuring shadowsocks-libev-ss-redir.
Configuring shadowsocksr-libev-ssr-redir.
Configuring tcping.
Configuring shadowsocks-libev-ss-server.
Configuring v2ray-core.
Configuring coreutils-nohup.
Configuring xray-core.
Configuring luci-app-passwall

zzduci, 2023-03-15:
以下为启动日志

Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000000] NX (Execute Disable) protection: active
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000000] SMBIOS 2.8 present.
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.0-0-gd239552ce722-prebuilt.qemu.org 04/01/2014
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000000] Hypervisor detected: KVM
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000002] kvm-clock: cpu 0, msr 23fe001, primary cpu clock
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000002] kvm-clock: using sched offset of 4844499813648 cycles
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000032] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.000058] tsc: Detected 2000.000 MHz processor
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004189] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004194] e820: remove [mem 0x000a0000-0x000fffff] usable
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.004210] last_pfn = 0x7ffde max_arch_pfn = 0x400000000
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004293] MTRR default type: write-back
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004302] MTRR fixed ranges enabled:
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004305]   00000-9FFFF write-back
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004307]   A0000-BFFFF uncachable
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004308]   C0000-FFFFF write-protect
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004309] MTRR variable ranges enabled:
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004312]   0 base 0080000000 mask FF80000000 uncachable
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004313]   1 disabled
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004314]   2 disabled
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004314]   3 disabled
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004315]   4 disabled
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004316]   5 disabled
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004317]   6 disabled
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.004318]   7 disabled
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.004362] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.018160] found SMP MP-table at [mem 0x000f5b70-0x000f5b7f]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019459] ACPI: Early table checksum verification disabled
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019636] ACPI: RSDP 0x00000000000F5990 000014 (v00 BOCHS )
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019662] ACPI: RSDT 0x000000007FFE28C6 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019694] ACPI: FACP 0x000000007FFE26A8 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019716] ACPI: DSDT 0x000000007FFDF040 003668 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019725] ACPI: FACS 0x000000007FFDF000 000040
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019732] ACPI: APIC 0x000000007FFE271C 000080 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019741] ACPI: SSDT 0x000000007FFE279C 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019750] ACPI: HPET 0x000000007FFE2866 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019758] ACPI: WAET 0x000000007FFE289E 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019766] ACPI: Reserving FACP table memory at [mem 0x7ffe26a8-0x7ffe271b]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019769] ACPI: Reserving DSDT table memory at [mem 0x7ffdf040-0x7ffe26a7]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019771] ACPI: Reserving FACS table memory at [mem 0x7ffdf000-0x7ffdf03f]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019774] ACPI: Reserving APIC table memory at [mem 0x7ffe271c-0x7ffe279b]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019776] ACPI: Reserving SSDT table memory at [mem 0x7ffe279c-0x7ffe2865]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019778] ACPI: Reserving HPET table memory at [mem 0x7ffe2866-0x7ffe289d]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.019781] ACPI: Reserving WAET table memory at [mem 0x7ffe289e-0x7ffe28c5]
Wed Mar 15 13:02:23 2023 kern.debug kernel: [    0.019894] ACPI: Local APIC address 0xfee00000
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.023485] Zone ranges:
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.023502]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.023558]   DMA32    [mem 0x0000000001000000-0x000000007ffddfff]
Wed Mar 15 13:02:23 2023 kern.info kernel: [    0.023562]   Normal   empty
[... cut ...]
Wed Mar 15 13:02:27 2023 cron.err crond[4226]: crond (busybox 1.33.2) started, log level 5
Wed Mar 15 13:02:28 2023 daemon.info dnsmasq[4289]: Connected to system UBus
Wed Mar 15 13:02:28 2023 cron.err crond[4587]: crond (busybox 1.33.2) started, log level 5
Wed Mar 15 13:02:28 2023 user.warn kernel: [   14.629068] passwall: restart when lan ifup
Wed Mar 15 13:02:28 2023 authpriv.info dropbear[4819]: Not backgrounding
Wed Mar 15 13:02:37 2023 user.info : luci: accepted login on / for root from 10.0.0.49

zzduci, 2023-03-15:
> 那就是你的依赖问题咯，你自己看日志啊

安装过程没有报依赖问题，下面有日志，都是官方原版初始状态

zzduci, 2023-03-15:
[image]
有趣的事，安装helloword想停止都停止不了，一直显示在运行，PASSWALL是想运行运行不成功，helloword停不下来

zzduci, 2023-03-15:
HELLOword，重启之后变成正常，PASSwall  还是一直未能正确启动

zzduci, 2023-03-15:
> 重启

估计是ipv6的问题，我把LAN口的IPV6全部禁用之后，重启PASSWALL可以正常工作了，给遇到同样问题的朋友一点经验，，现在的设置如下：
[image]

原来默认的设置为：

[image]

zzduci, 2023-03-15:
那么问题又来了，这个难道 不能用IPV6吗？这也不科学啊是吧

zzduci, 2023-03-15:
> 那么问题又来了，这个难道 不能用IPV6吗？这也不科学啊是吧

ipv6也可以用，但要先关掉安装好PASSWALL后再开启IPV6，可以用了

jh0072110, 2023-03-15:
……不要轰炸别人的邮箱谢谢