immortalwrt/immortalwrt #910: NanoPi R5S 固件存在问题

qingtian110, 2023-03-18:
### Describe the bug

拉取最新 master 分支源码，编译 nanopi-r5s 固件 **emmc 无法开机**，**tf 卡可以正常开机**，通过 tf 卡启动固件后，2.5G 网口**只能协商 1000M** 速率。

源码 commit `65a05816053836b0081ef830c25fd8e457813584`

这是我进行的编译操作，尚未对源码做任何更改：

```shell
git clone https://github.com/immortalwrt/immortalwrt --depth=1

cd immortalwrt/

./scripts/feeds update -a && ./scripts/feeds install -a

echo -e "CONFIG_TARGET_rockchip=y\nCONFIG_TARGET_rockchip_armv8=y\nCONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r5s=y\nCONFIG_PACKAGE_luci=y\nCONFIG_PACKAGE_luci-base=y" > .config

make defconfig

make -j17
```

无论我使用 dd 命令写入到 emmc 或者通过 瑞芯微开发工具 MASK 模式 `固件` + `rk356x_spl_loader_v1.15.112.bin` 方式写入，均无法启动，通电后电源红灯常亮。

------------

一下是通过 tf 卡启动得到的 dmesg，不知道能不能找到 2.5G 无法协议的原因。

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.98 (nssl@debian) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r0-65a0581) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT Thu Mar 16 07:02:02 2023
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 64 MiB at 0x00000000e5000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33432 r8192 d32104 u73728
[    0.000000] pcpu-alloc: s33432 r8192 d32104 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 console=tty1 earlycon=uart8250,mmio32,0xfe660000 swiotlb=1 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3769848K/3930112K available (8320K kernel code, 2374K rwdata, 5508K rodata, 1664K init, 731K bss, 94728K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[... 510 lines cut ...]

### ImmortalWrt version

master分支，忘记记录这个字段具体内容了

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5S

### Image kind

Self-built image

### Additional info

[image]
[image]

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-03-18:
协商的问题：
https://transfer.sh/ku7HQu/100-Revert-device-add-support-for-configuring-device-link-spe.patch
放到 package/network/config/netifd/patches 下面

或者用 R8169 驱动。

至于你说的刷入 EMMC 无法开机，无法复现，需要你自己接 TTL 看下是什么情况。

qingtian110, 2023-03-18:
我将 https://github.com/immortalwrt/immortalwrt/tree/master/package/kernel/r8125 目录内容替换成第三方的驱动后 https://github.com/sbwml/package_kernel_r8125 。2.5G 目前协商正常了，虽然不知道为什么显示为 2500baseX ，它明明是电口。不过它也工作正常了，iperf 测速也没有异常 2.37Gbps/s

```
Settings for eth2:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseX/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	                                     2500baseX/Full 
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	MDI-X: Unknown
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00000033 (51)
			       drv probe ifdown ifup
	Link detected: yes
```
[image]

不过EMMC 不能启动找到原因。也没有TTL 线在身边，目前有测试到 LEDE 的源码编译出来和第三方编译的OpenWrt 22.03 是可以正常通过 EMMC 启动的。

qingtian110, 2023-03-18:
> 至于你说的刷入 EMMC 无法开机，无法复现，需要你自己接 TTL 看下是什么情况。

好像uboot 的问题，我如果使用 LEDE 升级的方式将固件刷入 EMMC ，它可以正常启动。我只要完全破坏 EMMC 的数据后写入固件进去，他就无法启动了。

我是通过TF 启动固件，并将固件这样的方式写入EMMC
`dd if=immortalwrt-rockchip-armv8-friendlyarm_nanopi-r5s-squashfs-sysupgrade.img of=/dev/mmcblk1 bs=1M && sync`

qingtian110, 2023-03-19:
解决EMMC了，替换掉 arm-trusted-firmware-rockchip 和 uboot-rockchip 后在我的机器上启动了

closed 2023-03-19 by qingtian110