immortalwrt/immortalwrt #914: AX3200 5ghz

triatomic, 2023-03-19:
### Describe the bug

Hi, on my ax3200 on previous versions of immortalwrt i could use 160mhz width channel. 
Now the max i can use is 80mhz

### ImmortalWrt version

r25391-f5af73bcef

### ImmortalWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

triatomic, 2023-03-19:
Fixed with r25509-dcc8c8beaf

1715173329, 2023-03-19:
You should upgrade your firmware.

closed 2023-03-19 by 1715173329