immortalwrt/immortalwrt #918: Unable to dispatch: /cgi-bin/luci/admin/network/wifi

0neday, 2023-03-28:
### Describe the bug

404 Not Found
Sorry, the object you requested was not found.

No page is registered at '/admin/network'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

Unable to dispatch: /cgi-bin/luci/admin/network/wifi

### ImmortalWrt version

21.02

### ImmortalWrt target/subtarget

21.02

### Device

mt-3000

### Image kind

Self-built image

### Steps to reproduce

```
404 Not Found
Sorry, the object you requested was not found.

No page is registered at '/admin/network'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

Unable to dispatch: /cgi-bin/luci/admin/network/wifi
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-03-28:
你的构建里面带无线了吗？

1715173329, 2023-03-29:
那你的无线设置页面是 /cgi-bin/luci/admin/network/wireless 么？turboacc 只适配 luci-app-mtk 的无线设置页面。

0neday, 2023-03-29:
是的， where is luci-app-mtk

0neday, 2023-03-29:
thank you very much