immortalwrt/immortalwrt #928: nanopi r2s缺少驱动

azgh, 2023-04-07:
### Describe the bug

r2s是双千兆网卡，其中一个是usb3转千兆网卡，驱动正常应该是双千兆。
但由于目前的固件均缺乏网卡和usb3的驱动，导致显示/实际的网卡均为百兆。
请更新r2s的驱动，已实现r2s的双千兆网卡正确完整驱动。
谢谢。

### ImmortalWrt version

r19951-da9ffd2add

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R2S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-04-07:
垃圾网线或者垃圾设备的问题。

closed 2023-04-07 by 1715173329

azgh, 2023-05-20:
@1715173329 更换了网线，恢复了千兆，确实是网线太垃圾了，感谢。