immortalwrt/immortalwrt #935: MT7620 can not complie in Openwrt 21.02

trananh29, 2023-04-22:
### Describe the bug

make[4]: Entering directory '/home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.224'
  CC [M]  /home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2/core/mhi_init.o
In file included from /home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2/core/mhi_init.c:15:
./arch/mips/include/asm/uaccess.h: In function 'eva_kernel_access':
./arch/mips/include/asm/uaccess.h:90:9: error: implicit declaration of function 'uaccess_kernel'; did you mean 'die_if_kernel'? [-Werror=implicit-function-declaration]
  return uaccess_kernel();
         ^~~~~~~~~~~~~~
         die_if_kernel
/home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2/core/mhi_init.c: In function 'mon_text_read_u':
/home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2/core/mhi_init.c:563:24: warning: format '%llx' expects argument of type 'long long unsigned int', but argument 7 has type 'dma_addr_t' {aka 'unsigned int'} [-Wformat=]
      "%u %c %03d WP:%llx TRE: %llx %08x %08x",
                     ~~~^
                     %x
      ep->tstamp, ep->type, ep->chan, ep->wp,
                                      ~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:262: /home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2/core/mhi_init.o] Error 1
make[4]: *** [Makefile:1739: /home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2] Error 2
make[4]: Leaving directory '/home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.224'
make[3]: *** [Makefile:50: /home/a/Desktop/immortalwrt-v21/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/pcie_mhi-3.2/.built] Error 2
make[3]: Leaving directory '/home/a/Desktop/immortalwrt-v21/package/quectel_MHI'
time: package/quectel_MHI/compile#1.25#0.39#1.50
    ERROR: package/quectel_MHI failed to build.
make[2]: *** [package/Makefile:116: package/quectel_MHI/compile] Error 1
make[2]: Leaving directory '/home/a/Desktop/immortalwrt-v21'
make[1]: *** [package/Makefile:110: /home/a/Desktop/immortalwrt-v21/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/a/Desktop/immortalwrt-v21'
make: *** [/home/a/Desktop/immortalwrt-v21/include/toplevel.mk:230: world] Error 2

### ImmortalWrt version

21.02

### ImmortalWrt target/subtarget

mt7620

### Device

xiaomi mini

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-04-22:
Your local tree is out of date.

1715173329, 2023-04-22:
looks like you were compiling some 3rd patry packages, should ask the author.

closed 2023-04-22 by 1715173329