immortalwrt/immortalwrt #937: Runtime error (xxx.xxx.xxx.xxx/cgi-bin/luci/admin/docker/overview)

M33FT, 2023-04-26:
### Describe the bug

[image]

Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/model/cbi/dockerman/overview.lua:68: bad argument #1 to 'ipairs' (table expected, got nil)

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:794:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:795:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:998:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

### ImmortalWrt version

r25947-4227e7e46a

### ImmortalWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-04-26:
装 luci-compat 了吗？anyway 不建议在 openwrt 上面跑 docker

M33FT, 2023-04-26:
CONFIG_PACKAGE_luci-compat=y
Check [This ](https://paste.ofcode.org/775VuBvjfCnD8HDmtcT2Cw)
I had Docker in previous versions. I just wanted to let you know. Thanks bro