immortalwrt/immortalwrt #943: master编译失败

jh0072110, 2023-05-13:
### Describe the bug

X86编译失败

### ImmortalWrt version

master

### ImmortalWrt target/subtarget

无

### Device

X86

### Image kind

Self-built image

### Steps to reproduce

我直接贴错误日志吧…
FAILED: tests/modetest/modetest.p/modetest.c.o 
x86_64-openwrt-linux-musl-gcc -Itests/modetest/modetest.p -Itests/modetest -I../tests/modetest -I. -I.. -Itests -I../tests -I../include/drm -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c11 -include config.h -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/libdrm-2.4.115=libdrm-2.4.115 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -pthread -Wsign-compare -Werror=undef -Werror=implicit-function-declaration -Wpointer-arith -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wnested-externs -Wpacked -Wswitch-enum -Wmissing-format-attribute -Wstrict-aliasing=2 -Winit-self -Winline -Wshadow -Wdeclaration-after-statement -Wold-style-definition -Wno-unused-parameter -Wno-attributes -Wno-long-long -Wno-missing-field-initializers -Wno-pointer-arith -MD -MQ tests/modetest/modetest.p/modetest.c.o -MF tests/modetest/modetest.p/modetest.c.o.d -o tests/modetest/modetest.p/modetest.c.o -c ../tests/modetest/modetest.c
make[3]: *** [Makefile:81: /workdir/openwrt/build_dir/target-x86_64_musl/libdrm-2.4.115/.built] Error 1
In file included from ../tests/modetest/modetest.c:52:
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify/poll.h:42:30: error: "_REDIR_TIME64" is not defined, evaluates to 0 [-Werror=undef]
   42 | #if defined(_GNU_SOURCE) && !_REDIR_TIME64
      |                              ^~~~~~~~~~~~~
In function 'set_plane',
    inlined from 'set_planes' at ../tests/modetest/modetest.c:1698:7,
    inlined from 'main' at ../tests/modetest/modetest.c:2257:5:
../tests/modetest/modetest.c:1306:22: warning: 'ovr' may be used uninitialized [-Wmaybe-uninitialized]
 1306 |         ovr->crtc_id = p->crtc_id;
      |         ~~~~~~~~~~~~~^~~~~~~~~~~~
../tests/modetest/modetest.c: In function 'main':
../tests/modetest/modetest.c:1237:23: note: 'ovr' was declared here
 1237 |         drmModePlane *ovr;
      |                       ^~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libdrm'
time: package/feeds/packages/libdrm/compile#0.76#0.06#0.81
    ERROR: package/feeds/packages/libdrm failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/libdrm/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:114: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
以下为完整日志：
[11_Compile the firmware.txt](https://github.com/immortalwrt/immortalwrt/files/11469124/11_Compile.the.firmware.txt)

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-05-13:
https://github.com/openwrt/openwrt/pull/12590

1715173329, 2023-05-14:
upstream fixed

closed 2023-05-14 by 1715173329

ChrisLiu0343, 2023-05-29:
immortalwrt/build_dir/target-x86_64_musl/libdrm-2.4.115/.built] Error 1