immortalwrt/immortalwrt #948: qihoo 360 t7 dts 错误，wan口不工作

listenfree, 2023-05-17:
### Describe the bug

target/linux/mediatek/dts/mt7981-qihoo-360-t7.dtsi
...
		port@4 {
			reg = <4>;
			label = "wan";
		};
应该为
reg = <3>

### ImmortalWrt version

mediatek/filogic

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

qihoo 360 t7

### Image kind

Self-built image

### Steps to reproduce

qihoo 360 t7 dts 错误，wan口不工作

### Actual behaviour

target/linux/mediatek/dts/mt7981-qihoo-360-t7.dtsi
...
		port@4 {
			reg = <4>;
			label = "wan";
		};
应该为
reg = <3>

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-05-17:
这样说应该是
```dts
...
port@3 {
	reg = <3>;
...
```
？

listenfree, 2023-05-17:
我不懂，只是把reg 改到3以后，wan就ok了，也许改成port@3更好。

listenfree, 2023-05-17:
我改成port 3 reg <3>测试
刷的自编译immortalwrt-mediatek-filogic-qihoo_360-t7-ubootmod-squashfs-sysupgrade.bin,保留配置，保留了vlan设定等，测试ok。

1715173329, 2023-05-17:
感谢测试

closed 2023-05-17 by 1715173329

ningsibuqu, 2023-05-17:
想问一下，我使用 237 编译的 immortalwrt 固件和官方 immortalwrt 固件对比（快照+luci 而已）
前者测速可以到 800+ 后者同地点 WiFi 测速只能到 400M，是千兆宽带。
是我操作出了问题还是说无线驱动方面的差异导致？后续更新有可能满速吗？谢谢大佬了。

1715173329, 2023-05-18:
https://github.com/immortalwrt/lean-lede/commit/1cd432ab19a2eadad14c7f8e86cdebbb96e6a40f#diff-7bf8225b19a967d1c846f75831c2adefe6f143b257749ba5b74cc9f3624bb070R245
可以自己开下 wed 试试看咯

ningsibuqu, 2023-05-18:
> [immortalwrt/lean-lede@1cd432a#diff-7bf8225b19a967d1c846f75831c2adefe6f143b257749ba5b74cc9f3624bb070R245](https://github.com/immortalwrt/lean-lede/commit/1cd432ab19a2eadad14c7f8e86cdebbb96e6a40f#diff-7bf8225b19a967d1c846f75831c2adefe6f143b257749ba5b74cc9f3624bb070R245) 可以自己开下 wed 试试看咯

好了不少 可能有个 10M 以内的差距？谢谢L

commit 4ed52d78 (2023-05-17): mediatek: qihoo 360-t7: fix wan port reg val

Fixes: #948