immortalwrt/immortalwrt #957: zte_e8820s稳定版重启后丢配置

13528943945, 2023-05-24:
### Describe the bug

故障复现：
1.zte8820s刷入bread
2.bread界面刷入immortalwrt-21.02.6-ramips-mt7621-zte_e8820s-initramfs-kernel.bin
3.openwrt界面刷入升级包immortalwrt-21.02.6-ramips-mt7621-zte_e8820s-squashfs-sysupgrade.bin
4.openwrt界面修改配置，重启后配置丢失

曲线救国，先刷hc5962固件，不能忍受只有三个LAN口，于是再强制升级e8820s固件
1.zte8820s刷入bread
2.bread界面刷入immortalwrt-21.02.6-ramips-mt7621-hiwifi_hc5962-initramfs-kernel.bin
3.openwrt界面刷入升级包immortalwrt-21.02.6-ramips-mt7621-hiwifi_hc5962-squashfs-sysupgrade.bin
4.openwrt界面强制刷入升级包immortalwrt-21.02.6-ramips-mt7621-zte_e8820s-squashfs-sysupgrade.bin
5.openwrt界面修改配置，重启后配置生效

是个案还是通病，是kernel底包问题还是sysupgrade升级包问题，不得而知

### ImmortalWrt version

immortalwrt-21.02.6

### ImmortalWrt target/subtarget

ramips-mt7621

### Device

zte_e8820s

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-05-24:
Ping @SVSWMW

SVSWMW, 2023-05-24:
在bread界面直接刷入immortalwrt-21.02.6-ramips-mt7621-zte_e8820s-squashfs-factory.bin试试

13528943945, 2023-05-25:
immortalwrt-21.02.6-ramips-mt7621-zte_e8820s-squashfs-factory.bin

刷好bread后我第一时间就刷的它，机子反复重启，才开始折腾之旅

李云海

***@***.***

13528943945

电子名片新出VIP模板啦，快来体验>>

扫一扫,

快速添加名片到手机

--------------------------------------------------------------------------------

13528943945, 2023-06-04:
[    0.626842] mt7621-nand 1e003000.nand: Using programmed access timing: 31c07388
[    0.634423] nand: device found, Manufacturer ID: 0xc2, Chip ID: 0xf1
[    0.640774] nand: Macronix MX30LF1G18AC
[    0.644618] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    0.652180] mt7621-nand 1e003000.nand: ECC strength adjusted to 4 bits
[    0.658736] mt7621-nand 1e003000.nand: Using programmed access timing: 21005134
[    0.666038] mt7621-nand 1e003000.nand: Using programmed access timing: 21005134
[    0.673337] Scanning device for bad blocks
[    0.714857] Bad eraseblock 29 at 0x0000003a0000
[    0.935608] Bad eraseblock 200 at 0x000001900000
[    1.419275] Bad eraseblock 578 at 0x000004840000
[    1.906762] Bad eraseblock 959 at 0x0000077e0000
[    1.941197] Bad eraseblock 983 at 0x000007ae0000
[    1.996669] 7 fixed-partitions partitions found on MTD device mt7621-nand
[    2.003459] Creating 7 MTD partitions on "mt7621-nand":
[    2.008692] 0x000000000000-0x000000080000 : "u-boot"
[    2.015047] 0x000000080000-0x000000100000 : "debug"
[    2.021181] 0x000000100000-0x000000140000 : "factory"
[    2.027723] 0x000000140000-0x000000540000 : "kernel"
[    2.033947] 0x000000540000-0x0000021c0000 : "ubipart0"
[    2.040599] 0x0000021c0000-0x000002240000 : "bdinfo"
[    2.046809] 0x000002240000-0x000007f80000 : "ubipart1"
[    2.054367] Concatenating MTD devices:
[    2.058165] (0): "ubipart0"
[    2.060962] (1): "ubipart1"
[    2.063764] into device "ubi-concat"
[    2.067389] 1 fixed-partitions partitions found on MTD device ubi-concat
[    2.074085] Creating 1 MTD partitions on "ubi-concat":
[    2.079232] 0x000000000000-0x0000079c0000 : "ubi"
[    2.114033] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module

是否因为本机闪存有坏块而出现我上述的故障？bread是最新e8820s专用版，刷squashfs-factory.bin固件时，闪存布局无法选自动，只能手动指定极路由4（B70）。刷入后不断重启。

qzydustin, 2023-06-04:
小米路由器nano，breed刷21.02.6底包，openwrt刷21.02.6升级包也是这种情况，然后openwrt界面再刷入23.05升级包后就好了。建议直接刷23.05。

另外，21.02.6 openwrt界面备份配置，备份出来的0kb的配置文件。可能无法保存配置有关。

13528943945, 2023-06-06:
我目前只能两步操作才能正常运行：
1.BREAD下先刷入HC5962(B70)的initramfs-kernel.bin
2.openwrt下强制刷入E8820S的squashfs-sysupgrade.bin

以下两种方法均尝试过，失败告终：
1.BREAD下直接刷入E8820S的squashfs-factory.bin
结果：反复重启
2.先BREAD下刷入E8820S的initramfs-kernel.bin，再openwrt下刷入E8820S的squashfs-sysupgrade.bin
结果：如题无法保存配置，机器始终运行在initramfs模式。

1715173329, 2023-11-09:
https://github.com/immortalwrt/immortalwrt/commit/85e0f369a080d38b2c4728c8048f10a6c8938cd9

最近在 23.05+ 重做了一下对这个设备的支持，编译了专属 uboot，测试过一切正常。
最好是恢复一下原有分区 indiv（->config）和 tag（->factory）的备份。

closed 2023-11-09 by 1715173329

heygo1345678, 2025-12-04:
> [85e0f36](https://github.com/immortalwrt/immortalwrt/commit/85e0f369a080d38b2c4728c8048f10a6c8938cd9)
> 
> 最近在 23.05+ 重做了一下对这个设备的支持，编译了专属 uboot，测试过一切正常。 最好是恢复一下原有分区 indiv（->config）和 tag（->factory）的备份。
> 
> 21.02 因为 eol 和兼容问题，就不打算再改动了。

我 breed 刷 21.02.7 ，改好mac地址 5G信号正常的 25dbm，然后breed备份好这个信号正常的eeprom ，
根据专属 天灵uboot 刷 24.10.4 ， 5G 信号变成了 3dbm
于是24.10下面刷入这个信号正常的 eeprom，5G信号还是3dbm
root@E8820S:~# insmod mtd-rw.ko i_want_a_brick=1
root@E8820S:~# mtd write /tmp/eeprom.bin factory
Unlocking factory ...

Writing from /tmp/eeprom.bin to factory ... 

现在只能停留在 23.05 rc4 这个最后支持 breed uboot 版本下，这个版本下5G信号正常，因为是 breed支持的 eeprom
但是这个版本在线构建和 自己 imagebuilder 看好像都删除的差不多了。

heygo1345678, 2025-12-04:
> ### Describe the bug
> 故障复现： 1.zte8820s刷入bread 2.bread界面刷入immortalwrt-21.02.6-ramips-mt7621-zte_e8820s-initramfs-kernel.bin [...]

你曲线救国使用 hc5962固件 做过渡固件我用在了 天灵uboot + immortalwrt 24.10.4 降级到 breed + immortalwrt 23.05 rc4 上面了， 也会有这个症状， 用这个过渡下就好了， immortalwrt 23.05 早期几个支持 breed 分区布局的应该都删除了。

1715173329, 2025-12-04:
不支持 breed 修改的 eeprom，也不支持恩山那些所谓的魔改版。
你如果会基本的 mtd 读写操作，可以自己重新把备份的 e2p 按 offset 写入到对应的分区地址。