immortalwrt/immortalwrt #960: 中继BUG。固件中继模式下，开机不能自动连接中继上。

iamyangyiok, 2023-05-30:
### Describe the bug

在小米WR30U，红米AX6000测试同样问题，用5G中继的上节点信号，作接收端。当时中继连上可以用，但是重启并不会连接，也填了信号为0，自己搜寻，重启信道变了，且扫瞄也找不到，中继失败。固件用好用 ，望 能修复，感谢

### ImmortalWrt version

18.04

### ImmortalWrt target/subtarget

1

### Device

1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-05-30:
我们从未支持过 WR30U，也没有 18.04 的版本号。你是用的是第三方固件，请自行联系第三方发布者。

closed 2023-05-30 by 1715173329

iamyangyiok, 2023-05-30:
> 我们从未支持过 WR30U，也没有 18.04 的版本号。你是用的是第三方固件，请自行联系第三方发布者。

不好意思  版本我可能记错，那用你们红米AX6000快照固件也是这样的问题，中继重启不能自动连。

1715173329, 2023-05-30:
如果 MT76 有问题，请带上日志发送到 OpenWrt 上游。