immortalwrt/immortalwrt #961: Xiaomi Mi Router AC2100 UPnP not working

9thChasingWindGirl, 2023-05-30:
### Describe the bug

**_firewall4 + nftables + luci-app-upnp(nftables)_**

but not working,check at miniupnp

### ImmortalWrt version

r26451-84ddb9fecc

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router AC2100

### Image kind

Official downloaded image

### Expected behaviour

**UPnP working**

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-05-30:
If your router is behind NAT, I'd say it's just somehow expected.

9thChasingWindGirl, 2023-05-30:
> If your router is behind NAT, I'd say it's just somehow expected.

what's "behind nat",should i post some screenshot or log.txt?

1715173329, 2023-05-30:
> what's "behind nat"

No public IP available

liyiranlab, 2023-12-17:
用miniupnpd，光猫开启DMZ到主路由，/etc/config/upnpd 的config upnpd 'config' 下增加option external_ip '随意一个公网IP'，可开启upnp转发，例如option external_ip '120.232.145.144'

9thChasingWindGirl, 2023-12-17:
> 用miniupnpd，光猫开启DMZ到主路由，/etc/config/upnpd 的config upnpd 'config' 下增加option external_ip '随意一个公网IP'，可开启upnp转发，例如option external_ip '120.232.145.144'

感谢分享，已经退回到iptables了，nf和其他插件的兼容性有点问题