immortalwrt/immortalwrt #962: Issues: /etc/config/openssl config error

hu3rror, 2023-06-01:
### Describe the bug

uci show: `uci: Parse error (invalid command) at line 3, byte 0`

```
root@ImmotalWrt:~# cat /etc/config/openssl
config engine 'devcrypto'
        option enabled '1'
engine 'devcrypto'
        option enabled '1'
        option builtin '1'
```

This bug seems to be caused here?
https://github.com/immortalwrt/immortalwrt/blob/53ac682430017046a59478ea7141702ec25e78bf/package/libs/openssl/Makefile#L148-L160

### ImmortalWrt version

r26482-2464d48e17

### ImmortalWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-01:
It should be the cause: 0b70d55a64c39da4144cc15aa7d6de70a3957ccb
Upstream didn't notice it as they disabled built-in engine by default.

Will do a fix later.

1715173329, 2023-06-01:
Fixed.

closed 2023-06-01 by 1715173329