immortalwrt/immortalwrt #963: R4S

skaterjuice, 2023-06-01:
### Describe the bug

Once compiled, R4S LAN and WAN ports are unresponsive. WAN sometimes will have activity but LAN will not. Unable to access OpenWRT.

### ImmortalWrt version

23.05

### ImmortalWrt target/subtarget

R4S

### Device

NanoPi R4S

### Image kind

Self-built image

### Steps to reproduce

Compiled multiple versions of R4S, custom and default.

### Actual behaviour

WAN and LAN unresponsive. LEDS on the device show Power, System, WAN (sometimes) LAN off

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-01:
_Troubleshooting any problem without the error log is like driving with your eyes closed.
  -- [Apache Getting Started](https://httpd.apache.org/docs/2.4/getting-started.html)_

skaterjuice, 2023-06-01:
> 

I’m not certain how to get an error log when the compile completes with no errors and the image flashes with no errors. How do I get a log?

1715173329, 2023-06-01:
Catch the log via TTL/UART or try official built image

skaterjuice, 2023-06-01:
> Catch the log via TTL/UART or try official built image

Ok I’ll see If I can do that. Thank you.

closed 2023-06-17 by 1715173329