immortalwrt/immortalwrt #964: ERROR: package/boot/grub2 failed to build

carrotSun, 2023-06-01:
### Describe the bug

make[3]: Entering directory '/home/rabbit/immortalwrt/package/boot/grub2'
rm -f /home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/.built
touch /home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fhonour-copts -fmacro-prefix-map=/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06=grub-2.06 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fhonour-copts -fmacro-prefix-map=/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06=grub-2.06 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include " LDFLAGS="-L/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -Wl,--gc-sections -znow -zrelro " make -j1 -C /home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/grub-core AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fhonour-copts -fmacro-prefix-map=/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06=grub-2.06 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/grub-core'
make  all-am
make[5]: Entering directory '/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/grub-core'
x86_64-openwrt-linux-musl-gcc -std=gnu99 -Os -m32 -Wall -W -Wshadow -Wpointer-arith -Wundef -Wchar-subscripts -Wcomment -Wdeprecated-declarations -Wdisabled-optimization -Wdiv-by-zero -Wfloat-equal -Wformat-extra-args -Wformat-security -Wformat-y2k -Wimplicit -Wimplicit-function-declaration -Wimplicit-int -Wmain -Wmissing-braces -Wmissing-format-attribute -Wmultichar -Wparentheses -Wreturn-type -Wsequence-point -Wshadow -Wsign-compare -Wswitch -Wtrigraphs -Wunknown-pragmas -Wunused -Wunused-function -Wunused-label -Wunused-parameter -Wunused-value  -Wunused-variable -Wwrite-strings -Wnested-externs -Wstrict-prototypes -g -Wredundant-decls -Wmissing-prototypes -Wmissing-declarations  -Wextra -Wattributes -Wendif-labels -Winit-self -Wint-to-pointer-cast -Winvalid-pch -Wmissing-field-initializers -Wnonnull -Woverflow -Wvla -Wpointer-to-int-cast -Wstrict-aliasing -Wvariadic-macros -Wvolatile-register-var -Wpointer-sign -Wmissing-include-dirs -Wmissing-prototypes -Wmissing-declarations -Wformat=2 -march=i386 -mrtd -mregparm=3 -falign-jumps=1 -falign-loops=1 -falign-functions=1 -freg-struct-return -mno-mmx -mno-sse -mno-sse2 -mno-sse3 -mno-3dnow -msoft-float -fno-dwarf2-cfi-asm -mno-stack-arg-probe -fno-asynchronous-unwind-tables -fno-unwind-tables -fno-ident -fno-stack-protector -Wtrampolines   -ffreestanding  -Os -pipe -fno-caller-saves -fhonour-copts -fmacro-prefix-map=/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06=grub-2.06 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -m32 -Wl,-melf_i386 -Wl,--build-id=none  -nostdlib -Wl,-N -Wl,-r,-d  -L/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -Wl,--gc-sections -znow -zrelro  -o disk.module lib/disk_module-disk.o
/home/rabbit/immortalwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: --gc-sections requires a defined symbol root specified by -e or -u
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:27665: disk.module] Error 1
make[5]: Leaving directory '/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/grub-core'
make[4]: *** [Makefile:26916: all] Error 2
make[4]: Leaving directory '/home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/grub-core'
make[3]: *** [Makefile:182: /home/rabbit/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-pc/grub-2.06/.built] Error 2
make[3]: Leaving directory '/home/rabbit/immortalwrt/package/boot/grub2'
time: package/boot/grub2/pc/compile#0.31#0.02#0.28
    ERROR: package/boot/grub2 failed to build (build variant: pc).
make[2]: *** [package/Makefile:120: package/boot/grub2/compile] Error 1
make[2]: Leaving directory '/home/rabbit/immortalwrt'
make[1]: *** [package/Makefile:114: /home/rabbit/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rabbit/immortalwrt'
make: *** [/home/rabbit/immortalwrt/include/toplevel.mk:231: world] Error 2

### ImmortalWrt version

master

### ImmortalWrt target/subtarget

-

### Device

x86

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-01:
We cannot reproduce this error on our buildbot, you may probably need to check your build environment or run `make dirclean`

carrotSun, 2023-06-01:
> 

Maybe related with the kernel(version 5.15.114) ? 
When I use kernel 5.15.113, everything is ok.

1715173329, 2023-06-01:
So I guess you enabled `CONFIG_USE_GC_SECTIONS` option, but unfortunately grub2 doesn't seem to play nice with it.
To make it compile, add `no-gc-sections` flag to grub2's `PKG_BUILD_FLAGS` variable.

carrotSun, 2023-06-02:
> So I guess you enabled `CONFIG_USE_GC_SECTIONS` option, but unfortunately grub2 doesn't seem to play nice with it. To make it compile, add `no-gc-sections` flag to grub2's `PKG_BUILD_FLAGS` variable.

Thanks!  I will try it later.

closed 2023-06-11 by 1715173329