immortalwrt/immortalwrt #965: error: pcre.h: No such file or directory

xiaobaishu1, 2023-06-01:
### Describe the bug

mkdir -p /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
echo '7' | cmp -s - /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libltdl.version || { echo '7' > /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libltdl.version;  } 
SHELL= flock /home/openwrt/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/libtool-2.4.7/.pkgdir/libltdl/. /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libltdl_installed
echo "libltdl" >> /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libtool.default.install
make[3]: Leaving directory '/home/openwrt/openwrt/package/libs/libtool'
time: package/libs/libtool/compile#0.13#0.03#0.15
make[3]: Entering directory '/home/openwrt/openwrt/package/libs/readline'
mkdir -p /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
echo '8' | cmp -s - /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libreadline.version || { echo '8' > /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libreadline.version;  } 
SHELL= flock /home/openwrt/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/readline-8.2/.pkgdir/libreadline/. /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libreadline_installed
echo "libreadline" >> /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/readline.default.install
make[3]: Leaving directory '/home/openwrt/openwrt/package/libs/readline'
time: package/libs/readline/compile#0.15#0.06#0.20
make[3]: Entering directory '/home/openwrt/openwrt/package/utils/lua5.3'
mkdir -p /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
echo '5.3' | cmp -s - /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/liblua5.3.version || { echo '5.3' > /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/liblua5.3.version;  } 
SHELL= flock /home/openwrt/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/lua-5.3.5/.pkgdir/liblua5.3/. /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.liblua5.3_installed
echo "liblua5.3" >> /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/lua5.3.default.install
make[3]: Leaving directory '/home/openwrt/openwrt/package/utils/lua5.3'
time: package/utils/lua5.3/compile#0.15#0.03#0.18
make[3]: Entering directory '/home/openwrt/openwrt/feeds/packages/net/haproxy'
rm -f /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13/.built
touch /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13/.built_check
make TARGET=linux-musl -C /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13 DESTDIR="/home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13/ipkg-install" CC="aarch64-openwrt-linux-musl-gcc" PCREDIR="/home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/usr/" USE_LUA=1 LUA_LIB_NAME="lua5.3" LUA_INC="/home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/lua5.3" LUA_LIB="/home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib" SMALL_OPTS="-DBUFSIZE=16384 -DMAXREWRITE=1030 -DSYSTEM_MAXCONN=165530" USE_ZLIB=1 USE_PCRE=1 USE_PCRE_JIT=1 USE_PTHREAD_PSHARED=1 USE_LIBATOMIC=1 USE_PROMEX=1 VERSION="2.6.13" SUBVERS="-1" VERDATE="2023/06/01" IGNOREGIT=1 USE_OPENSSL=1 ADDLIB="-lcrypto -lm" CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13=haproxy-2.6.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fno-strict-aliasing -Wdeclaration-after-statement -Wno-unused-label -Wno-sign-compare -Wno-unused-parameter -Wno-clobbered -Wno-missing-field-initializers -Wno-cast-function-type -Wno-address-of-packed-member -Wtype-limits -Wshift-negative-value -Wshift-overflow=2 -Wduplicated-cond -Wnull-dereference -fwrapv -fasynchronous-unwind-tables -Wno-null-dereference" LD="aarch64-openwrt-linux-musl-gcc" LDFLAGS="-L/home/openwrt/openwrt/staging_dir/toolchain-aarch64_generic_gcc-12.3.0_musl/usr/lib -L/home/openwrt/openwrt/staging_dir/toolchain-aarch64_generic_gcc-12.3.0_musl/lib -znow -zrelro"
make[4]: Entering directory '/home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13'
  CC      src/ssl_sample.o
In file included from include/haproxy/pattern-t.h:28,
                 from include/haproxy/acl-t.h:25,
                 from include/haproxy/acl.h:25,
                 from src/ssl_sample.c:22:
include/haproxy/regex-t.h:31:10: fatal error: pcre.h: No such file or directory
   31 | #include <pcre.h>
      |          ^~~~~~~~
compilation terminated.
make[4]: *** [Makefile:1017: src/ssl_sample.o] Error 1
make[4]: Leaving directory '/home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13'
make[3]: *** [Makefile:147: /home/openwrt/openwrt/build_dir/target-aarch64_generic_musl/haproxy-ssl/haproxy-2.6.13/.built] Error 2
make[3]: Leaving directory '/home/openwrt/openwrt/feeds/packages/net/haproxy'
time: package/feeds/packages/haproxy/ssl/compile#0.14#0.09#0.23
    ERROR: package/feeds/packages/haproxy failed to build (build variant: ssl).
make[2]: *** [package/Makefile:120: package/feeds/packages/haproxy/compile] Error 1
make[2]: Leaving directory '/home/openwrt/openwrt'
make[1]: *** [package/Makefile:114: /home/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/openwrt'
make: *** [/home/openwrt/openwrt/include/toplevel.mk:231：world] 错误 2

### ImmortalWrt version

master

### ImmortalWrt target/subtarget

无

### Device

电牛子R66S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

hong0980, 2023-06-01:
* pkg_hash_check_unresolved: cannot find dependency kmod-fs-ntfs3 for ntfs3-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for ntfs3-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 * 	kmod-fs-ntfs3
 * opkg_install_cmd: Cannot install package automount.
 * check_conflicts_for: The following packages conflict with ntfs3-mount:
 * check_conflicts_for: 	ntfs-3g * 
 * opkg_install_cmd: Cannot install package ntfs3-mount.
 我搭车问下openwrt-21.02，openwrt-23.05编译了几次都是冲突，就是这四五天的代码的问题，天灵大神帮忙解决下，谢谢。

1715173329, 2023-06-01:
Duplicate of immortalwrt/packages#746