immortalwrt/immortalwrt #974: 不是BUG！DHCP客户端的Vendor Class能否改进一下

13528943945, 2023-06-08:
### Describe the bug

最近爱快的新版本，dhcp协议，Vendor Class部分多了字符串及十六进制的选项，这极大方便了玩iptv（ipoe方式接入）的朋友。openwrt这边也可以ipoe，只不过需要改dhcp脚本，略显麻烦。能否借鉴爱快的方法？将抓包后的dhcp60的值（16进制）直接填入Vendor Class并指明16进制。这样不改/lib/netifd/proto/dhcp.sh脚本，就能获取到10开头的iptv内网地址

### ImmortalWrt version

all

### ImmortalWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-08:
PR is welcome

windybird, 2023-06-10:
这个地方要智能点就好了

ROOOCCCK, 2023-06-22:
请求增加NX30 PRO的编译选项，电报群一直没给我通过=-=

darescloud, 2023-07-23:
> PR is welcome

能否内核模块加入wdat_wdt.ko支持-esxi watchdog

ciegyet, 2024-02-26:
臣附议！