immortalwrt/immortalwrt #977: Dependencies for block-mount are missing

hu3rror, 2023-06-15:
### Describe the bug

```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20220927 for block-mount
 * pkg_hash_check_unresolved: cannot find dependency libblobmsg-json20220927 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 *      libubox20220927
 *      libblobmsg-json20220927
 * opkg_install_cmd: Cannot install package automount.
 * pkg_hash_check_unresolved: cannot find dependency libubox20220927 for mtd
 * pkg_hash_fetch_best_installation_candidate: Packages for mtd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mtd.
make[2]: *** [Makefile:186: package_install] Error 255
make[1]: *** [Makefile:141: _call_image] Error 2
make: *** [Makefile:259: image] Error 2
```

### ImmortalWrt version

r26503-a5699b283d

### ImmortalWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-15:
unfortunately libubox was updated 3 days ago (see c6531042da2a) and the ABI was changed due to it.
block-mount is a non-shared package so in 23.05-rc1 it's fixed and we cannot update it.

To workaround it, you can disable automount by `make image PACKAGES="-automount ..."` or just download official built image.

1715173329, 2023-06-17:
Let's reopen it for now - in case someone who runs into the same issue.
We will create -rc2 for it in the next few days.

closed 2023-06-22 by 1715173329