immortalwrt/immortalwrt #978: AX3600 stock刷机后无法启动

zlopen, 2023-06-15:
### Describe the bug

master刷机后无法启动，wifi和网线能连，但拿不到ip，静态设ip也无法连接
U-Boot 2016.01 (Feb 19 2020 - 10:39:20 +0000), Build: jenkins-r3600_ota_publish-25

DRAM:  smem ram ptable found: ver: 1 len: 4
512 MiB
NAND:  ONFI device found
ID = 1590aaef
Vendor = ef
Device = aa
SF: Unsupported flash IDs: manuf ff, jedec ffff, ext_jedec ffff
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:   sdhci: Node Not found, skipping initialization

PCI Link Intialized
PCI1 is not defined in the device tree
In:    serial@78B3000
Out:   serial@78B3000
Err:   serial@78B3000
machid: 8010010
MMC Device 0 not found
eth5 MAC Address from ART is not valid
write phy_id=1, reg(0x8074):0x0670
write phy_id=2, reg(0x8074):0x0670
write phy_id=3, reg(0x8074):0x0670
write phy_id=4, reg(0x8074):0x0670
bootwait is on, bootdelay=5
Hit any key to stop autoboot:  0
 restore_defaults is set, enlarge xqup detect time
 trigger button release!
boot from rootfs 1
  miwifi: check crash in rmem !
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: volume 2 ("rootfs_data") re-sized from 9 to 760 LEBs
ubi0: attached mtd1 (name "mtd=0", size 128 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 1024, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1686761402
ubi0: available PEBs: 0, total reserved PEBs: 1024, PEBs reserved for bad PEB handling: 40
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (6348800)
Could not find PCI in device tree
## Loading kernel from FIT Image at 44000000 ...
   Using 'config@ac04' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.1.33
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x440000e8
     Data Size:    6180515 Bytes = 5.9 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x41000000
     Entry Point:  0x41000000
     Hash algo:    crc32
     Hash value:   985b4314
     Hash algo:    sha1
     Hash value:   2fef51c573ab05f84e122a68aa8e78c4b87e962d
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 44000000 ...
   Using 'config@ac04' configuration
   Trying 'fdt-1' fdt subimage
     Description:  ARM64 OpenWrt xiaomi_ax3600-stock device tree blob
     Type:         Flat Device Tree
[... 447 lines cut ...]
r23381-3810066f16

### Device

Xiaomi AX3600 (stock layout)

### Image kind

Self-built image

### Steps to reproduce

基于master分支编译，stock默认配置基础上加了luci图形界面

### Actual behaviour

启动后wifi和网线可连，但拿不到ip

### Expected behaviour

能正常工作

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-15:
确实烂了，推荐先用 23.05 稳定分支

1715173329, 2023-06-16:
should be fixed by a90939c56327996b013e12cb60a5bb10af7fd57a

mitsukileung, 2023-06-16:
> should be fixed by [a90939c](https://github.com/immortalwrt/immortalwrt/commit/a90939c56327996b013e12cb60a5bb10af7fd57a)

[image]

[image]

大佬，这里的内核要不要改回 5.15 ？

1715173329, 2023-06-16:
草，改错了

colinhd8, 2023-06-16:
@1715173329 
大佬，是不是6.1内核的x64固件也挂了？
我用如下的配置(基本上可以认为是最小化编译了)
`CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DISPLAY_SUPPORT=y
CONFIG_PACKAGE_alsa-ucm-conf=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bnx2-firmware=y
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-hwmon-core=y
CONFIG_PACKAGE_kmod-i2c-algo-bit=y
CONFIG_PACKAGE_kmod-i2c-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-ledtrig-audio=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-net-selftests=y
CONFIG_PACKAGE_kmod-phy-ax88796b=y
CONFIG_PACKAGE_kmod-phylink=y
CONFIG_PACKAGE_kmod-pps=y
CONFIG_PACKAGE_kmod-ptp=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_libcomerr=y
CONFIG_PACKAGE_libext2fs=y
CONFIG_PACKAGE_libss=y
CONFIG_PACKAGE_ntfs3-mount=y
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_VMDK_IMAGES=y
`
编译后，在虚拟机中无法启动，在启动菜单选择一项进去后卡在这里了：
[image]
而同样的配置，用5.15内核就一切正常。

closed 2023-06-16 by 1715173329