immortalwrt/immortalwrt #980: master分支X64固件6.1内核无法启动

colinhd8, 2023-06-17:
### Describe the bug

master分支(截止目前最新代码)X64固件6.1内核在vmware虚拟机中无法启动，卡在这个界面：
[image]
同样的配置，切换为5.15内核后一切正常。
还请大佬帮忙分析下，谢谢。

### ImmortalWrt version

master b6d8a0e65e4d77d3dd48b09f375a6ef1ccb188f2

### ImmortalWrt target/subtarget

x64

### Device

x64 vmware

### Image kind

Self-built image

### Actual behaviour

卡在启动界面，无法启动。

### Expected behaviour

正常启动。

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DISPLAY_SUPPORT=y
CONFIG_PACKAGE_alsa-ucm-conf=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bnx2-firmware=y
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-hwmon-core=y
CONFIG_PACKAGE_kmod-i2c-algo-bit=y
CONFIG_PACKAGE_kmod-i2c-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-ledtrig-audio=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-net-selftests=y
CONFIG_PACKAGE_kmod-phy-ax88796b=y
CONFIG_PACKAGE_kmod-phylink=y
CONFIG_PACKAGE_kmod-pps=y
CONFIG_PACKAGE_kmod-ptp=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_libcomerr=y
CONFIG_PACKAGE_libext2fs=y
CONFIG_PACKAGE_libss=y
CONFIG_PACKAGE_ntfs3-mount=y
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_VMDK_IMAGES=y
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-17:
我们没有 kernel 6.1 x86 的支持，你的固件是从哪儿来的

colinhd8, 2023-06-17:
> 我们没有 kernel 6.1 x86 的支持，你的固件是从哪儿来的
手动更改target/linux/x86/Makefile中的KERNEL_PATCHVER:=6.1
然后编译的。那就是现在不管什么设备类型都还不支持6.1内核是吧。

1715173329, 2023-06-17:
那你得看提交记录有没有这个 target 的支持。

closed 2023-06-17 by 1715173329

colinhd8, 2023-06-17:
> 那你得看提交记录有没有这个 target 的支持。 或者 config-6.1

明白了。谢谢。

gbm777, 2024-01-31:
> 那你得看提交记录有没有这个 target 的支持。 或者 config-6.1

请问如何能config-6.1，请帮忙引路

xlighting2017, 2024-01-31:
> > 那你得看提交记录有没有这个 target 的支持。 或者 config-6.1
> 
> 请问如何能config-6.1，请帮忙引路

加入openwrt开发组，给代码库提交PR
不会也可以学
不学也可以等

colinhd8, 2024-01-31:
> > 那你得看提交记录有没有这个 target 的支持。 或者 config-6.1
> 
> 请问如何能config-6.1，请帮忙引路

master分支已经支持了。